autowarefoundation/autoware_universe #744: The dicision of vehicle state (vehicle is stopping or moving) should be judged by a common node.

satoshi-ota, 2022-04-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In current architecture, various nodes judge whether the vehicle is stopping or moving independently. For example, the following nodes has function to judge the vehicle state and the threshold parameters:

- planning/behavior_velocity_planner
- control/longitudinal_controller
- control/vehicle_cmd_gate

On the other hand, these function has no common criteria for stopping or moving. I think if the criteria are not commonized throughout the autoware, the possibility of unexpected malfunctions increases. Futhermore, the maintenance costs also increase due to double management of parameters.

### Purpose

The purpose of this task is to create node that judges whether the vehicle is stopping or moving and publishes the vehicle state to other nodes. In addition, I think that the elapsed time since the vehicle transitioned to the current status, because some nodes want to know that how many seconds have elapsed since the vehicle has stopped (e.g. `stop_line` module in behavior_velocity_planner).

### Possible approaches

- create new node
- create new topic type

### Definition of done

- [ ] Create node that judges and publishes the vehicle state (stopping or moving) with elapsed time since the vehicle transitioned to the current status.
- [ ] Replace functions defined in various nodes with topics output from the node that created in this task.

satoshi-ota, 2022-04-21:
@TakaHoribe @yukkysaito

JianKangEgon, 2022-04-22:
It's a good idea to monitor vehicle state by one united node. From road test of real vehicle on site, as far as I'm concerned, the localization kinematics status and vehicle status from driving-by-wire module are supposed to be taken in to account as well. If these motion data are aligned with each other or not.

takayuki5168, 2022-04-23:
@satoshi-ota 
We already have `stop_filter` package here. This is not used at all though.
https://github.com/autowarefoundation/autoware.universe/tree/main/localization/stop_filter

takayuki5168, 2022-04-23:
Currently velocity and angular velocity are considered. IMO, cceleration and angular acceleration must be considered as well, at least.

satoshi-ota, 2022-09-25:
@takayuki5168 For now, this issue seems to be low priority, so I'll close this. If necessary, I'll make this issue open again.

closed 2022-09-25 by satoshi-ota