autowarefoundation/autoware_universe #768: Keyboard input of joy_controller

takayuki5168, 2022-04-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

We can teleoperate the vehicle with a joy controller. Currently some types of joy controllers are supported though keyboard input is not supported.

### Purpose

To test teleoperation only with a laptop or desktop pc, without any joy controllers.

### Possible approaches

Add  keyboard input in the `joy_controller` package.

### Definition of done

We can teleoperate the vehicle with keyboard. We can check it in a simulation.

takayuki5168, 2022-04-23:
@kenji-miyake 
Does anyone have a good idea about how to get asynchronous keyboard inputs with C++? I feel OpenCV is the easiest to do that.

kenji-miyake, 2022-04-23:
@takayuki5168 Kimura-san tried it before, but didn't work so perfectly.
If you think OpenCV is good, you can give it a try!

takayuki5168, 2022-04-23:
Thank you for your information.
@tkimura4 If you have any knowledge about it, please let me know.

shmpwk, 2022-07-08:
I've already created the minimum feature, but it is easier to use rviz panel instead.
If there is a usecase which rviz panel cannot do, let me know, and I will implement it.

takayuki5168, 2022-07-14:
@shmpwk 
We can close this issue since there seems to be no need for keyboard input of joy_controller

shmpwk, 2022-10-17:
Related discussion
https://github.com/orgs/autowarefoundation/discussions/2951

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant