autowarefoundation/autoware_universe #796: Use /vehicle/status/velocity_status as the input for lat/lon controller instead of /localization/kinematic_state

JianKangEgon, 2022-04-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, the control module takes the kinematic states from localization module as the input. In other words, the performance of control module relies on the localization results by a large margin. In simulator, it works fine indeed, because the localization module and vehicle module are actually fake. 

However, during road test on real vehicle, the robustness of localization module performs less reliable as vehicle driving-by-wire module, or wheel odometry sensor for robot. From my understanding, the algorithms of localization module (ICP / NDT matcher) are iterate computation progress.  Sometimes, the steering wheel jitters due to the lost localization matcher. In comparison, the can bus data or odometer could provide relatively more helpful information.

By communication with the vehicle can bus or wheel encoder, the control module is able to acquire a much more accurate value. Thus, I'd love to propose this issue and discuss if it's reasonable to do so.

### Purpose

Read vehicle velocity from can bus (or wheel encoder of robot) if these information are available.

### Possible approaches

- Remap the input topic for control module from /localization/kinematic_state to /vehicle/status/velocity_status as the velocity source.
- Change the corresponding message type

### Definition of done

- Rewrite the node file to change the topic message type.
- Build successfully without any conflict
- Validation and verification in both planning simulator and road test on real vehicle

TakaHoribe, 2022-04-25:
Thank you for your post.

The can bus speed information (` /vehicle/status/velocity_status`) is passed to the localization module for proper processing. If the can bus information is clearly more accurate than the localization matching, you can tell the information to the localization module through the error covariance.

The localization module also handles cases where the speed information from the can bus is disconnected. In such cases, the speed is calculated from the position estimation results. Therefore, I would suggest tuning the covariance parameters in Twist message sent to the EKF instead of sending the can bus information directly to the control module.

JianKangEgon, 2022-04-25:
@TakaHoribe Thanks for your quick reply. This solves my concern.

plane-li, 2022-04-26:
> Thank you for your post.
>
> The can bus speed information (` /vehicle/status/velocity_status`) is passed to the localization module for proper processing. If the can bus information is clearly more [...]

I agree with you! However,sometimes, the output of localization will shake in the longitudinal direction,which will lead to small different in velocity between EKF and Can bus.  I think it may caused by the ndt when vehicle run on the road which has little variation in the longitudinal  direction.

Sharrrrk, 2022-05-16:
Refer to https://github.com/orgs/autowarefoundation/discussions/291#discussioncomment-2731408
Will close this issue.

closed 2022-05-16 by Sharrrrk