autowarefoundation/autoware_universe #798: obstacle_avoidance_planner dies when the interval of path points is long

kosuke55, 2022-04-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

**This is a problem that does not occur with the current hard code, but could potentially occur.**

truncated_points size will be `1` when the interval of points in path is less than [`5.0`](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/obstacle_avoidance_planner/src/node.cpp#L1384). This causes the node to die when it finally accesses [`seg_idx + 1`](https://github.com/autowarefoundation/autoware.universe/blob/main/common/tier4_autoware_utils/include/tier4_autoware_utils/trajectory/trajectory.hpp#L128)
The interval between paths of `behavior_velocity` is [`1.0` with interporate](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_velocity_planner/include/utilization/path_utilization.hpp#L31-L33), so this is implicitly avoided.

### Expected behavior

obstacle_avoidance_planner does not die with any interval of path points.

### Actual behavior

obstacle_avoidance_planner dies with interval longer than 5.0.

### Steps to reproduce

Set [interval of path points](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_velocity_planner/include/utilization/path_utilization.hpp#L31-L33) greater than [clip length](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/obstacle_avoidance_planner/src/node.cpp#L1384)

kosuke55, 2022-04-25:
@takayuki5168

mitsudome-r, 2022-06-14:
@kosuke55 @takayuki5168 Do you have any plan to fix this issue? (or do have any idea about the approach so that others can work on the implementation?)

kosuke55, 2022-06-15:
@mitsudome-r 
Since this does not currently occur with the current hard code, there are no plans to deal with this issue as a low priority.

closed 2022-08-19 by takayuki5168