autowarefoundation/autoware_universe #812: can not record the self-defined topic using ros2 bag record

plane-li, 2022-04-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I can record the image and pointcloud throught" ros2 bag record --topic",however can not record the self-defined topic like"/perception/object_recognition/detection/apollo/objects".
Before command" ros2 bag record",I have done"source install/setup.bash " in the worksapce. At tha same time,the autoware is running,the network is ok and  "ros2 topic hz" is also usefull.

### Expected behavior

ros2 bag can record any topic in the related workspace!

### Actual behavior

It is always in the state of " Listening for topics..."
[image]

### Steps to reproduce

1.roslaunch autoware
2.open another terminal 
3.source install/setup.bash
4.ros2 bag record /perception/object_recognition/detection/apollo/objects

BonoloAWF, 2022-04-26:
@kenji-miyake @angry-crab could you please check whether you are able to reproduce this bug? rosbag2 should be recording the available topics.

kenji-miyake, 2022-04-26:
@plane-li I have two requests. :pray: 
- Could you provide the rosbag if you use some log data?
- Could you write the exact commands you executed?

@BonoloAWF I can try to reproduce this if he can provide more information.
I have one question and one request for you.
- Is this related to some projects such as Bus ODD?
- I think it's good to assign only one person to avoid double work and to make the driver clear.

BonoloAWF, 2022-04-26:
@kenji-miyake  Noted, I've updated the assignees. It is indirectly related to the Bus ODD since this is a functionality we will need.

angry-crab, 2022-04-27:
@kenji-miyake Sure! I'll look into it.

plane-li, 2022-04-27:
Thank you very much!, We have solved it.
I have tried to use another computer to record the topic my computer pub, it is ok! 
So It may have some problems with the ros2 environment  of my computer.
Two steps to solve:
1.reinstall the rosbag.
   **command: sudo apt-get install ros-galactic-rosbag2***
2.rebuild the aw
  **command: colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release**
 
After the above two steps in my computer,  the ros2 bag can record available topics.

closed 2022-04-28 by Sharrrrk