autowarefoundation/autoware_universe #822: Steering angle limit reached in simulation

08beeqtufail-sudo, 2022-04-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello guys, 
I am testing this repository in simulation with the sample lanelet2 and pointcloud maps provided. As i try to run the simulation with planning_simulator.launch.xml, i can see that the steering angle displayed in rviz reaches a limit of 20 degrees in the curves and never goes beyond it. It feels like somewhere the steering limit is being set to 20 degrees. The effect is more visible if we change the vehicle info parameters (bigger vehicle). In this case, the vehicle goes completely out of lane before coming back because the steering angle is limited to 20 degrees and it needs to steer more for bigger vehicles in curves. Do you guys have any suggestions as what could be the possible cause?
 
Note: I have already verified the max_steer_deg value in obstacle_avoidance_planner module and it is 40 degrees. Also, i have deactivated both EB optimization and mpt optimization to see whether it is related to optimization but it doesn't change the behavior.

https://user-images.githubusercontent.com/60785554/165594957-28df4685-4168-4005-a47c-06d3a24ac873.mp4

### Expected behavior

In a sharp curve, steering angle should not be saturated at 20 degrees. 

### Actual behavior

The steering angle is saturated at 20 degrees

### Steps to reproduce

- Change the vehicle_info parameters as follows:
    wheel_radius: 0.39
    wheel_width: 0.42
    wheel_base: 3.682 # between front wheel center and rear wheel center
    wheel_tread: 1.750 # between left wheel center and right wheel center
    front_overhang: 0.869 # between front wheel center and vehicle front
    rear_overhang: 1.024 # between rear wheel center and vehicle rear
    left_overhang: 0.360 # between left wheel center and vehicle left
    right_overhang: 0.360 # between right wheel center and vehicle right
    vehicle_height: 2.499

- Run the planning simulator
- Give an initial pose and specify a goal pose such that the path has a sharp curve.
You will see that the steering angle is saturate to 20 degrees in sharp curves, which leads to the vehicle going out of its lane.

### Versions

- OS : Ubuntu 20.04
- ROS2: galactic
- Autoware: https://github.com/autowarefoundation/autoware/tree/main 
 The commit number b260f67085dda5c0161ade051b236a3fbdee4147  dated 29 March 2022

JianKangEgon, 2022-04-28:
@08beeqtufail-sudo Hi, although the video seems to be broken from my side. But you can just modify this value "steer_lim_deg" in the config file src/universe/autoware/launch/tier4_control_launch/config/trajectory_follower/lateral_controller.param.yaml

This parameter is to limit the output of lateral controller regarding steering angle. I've checked this param is currently updated to 40.0

08beeqtufail-sudo, 2022-04-28:
@JianKangEgon Thank you for your response. Indeed that was the problem. It's working as expected now. I will close this issue.

closed 2022-04-28 by 08beeqtufail-sudo