autowarefoundation/autoware_universe #825: Docker Compose integration

kasperornmeck, 2022-04-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

From https://docs.docker.com/compose/:
_Compose is a tool for defining and running multi-container Docker applications. With Compose, you use a YAML file to configure your application’s services. Then, with a single command, you create and start all the services from your configuration._

Integrate Docker Compose into Autoware.Universe to build container images.

### Purpose

Lay the ground work, with the right tool, to enable building Autoware.Universe as a microservices architecture.

### Possible approaches

Replace _autoware/docker/build.sh_ with a Docker Compose YAML configuration file.

### Definition of done

When "docker-compose build" can build the current Autoware container image.

kasperornmeck, 2022-05-10:
@LalithVipulananthan 
We would need a volunteer who could do the work for this issue, I'll be able to assist of course. From my knowledge, there's only one script which build the actual container image. The files can be found here:
- /home/ubuntu/autoware/docker/build.sh
- /home/ubuntu/autoware/docker/autoware-universe/Dockerfile
- /home/ubuntu/autoware/docker/autoware-universe/docker-bake.hcl

Replacing the _build.sh_ file would be the first step.

kenji-miyake, 2022-05-10:
@kasperornmeck cc @LalithVipulananthan 
FYI, as written in https://github.com/autowarefoundation/autoware/blob/main/docker/README.md, `build.sh` is for the local build.

The actual images hosted on https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe are created by these workflows and actions.
- https://github.com/autowarefoundation/autoware/blob/main/.github/workflows/build.yaml
- https://github.com/autowarefoundation/autoware/blob/main/.github/workflows/build-self-hosted.yaml
- https://github.com/autowarefoundation/autoware/blob/main/.github/actions/docker-build-and-push/action.yaml

> Replacing the build.sh file would be the first step.

Regarding this part, could you first start with creating it as another script like `build-compose.sh`?
And after checking the behavior and reviewing it, I'd like to consider replacing it.

kasperornmeck, 2022-05-18:
@kenji-miyake 
Thanks for sharing these links. I'll have to look more into these to understand the workflow in GitHub.

kenji-miyake, 2022-05-18:
@kasperornmeck No, they are not so different. The differences are just some settings.
To build Docker images with GitHub Actions, it's better to use useful third-party actions. But they can't be used in our local, so we need to change some settings to handle the differences.