autowarefoundation/autoware_universe #851: Resolve warnings of hdd_reader

kenji-miyake, 2022-05-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

```
--- stderr: system_monitor
In file included from /usr/include/string.h:535,
                 from /usr/include/c++/11/cstring:42,
                 from /usr/include/boost/serialization/library_version_type.hpp:18,
                 from /usr/include/boost/serialization/map.hpp:28,
                 from /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/include/hdd_reader/hdd_reader.hpp:23,
                 from /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/reader/hdd_reader/hdd_reader.cpp:20:
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘int get_ata_identify(int, HDDInfo*)’ at /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/reader/hdd_reader/hdd_reader.cpp:223:10:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ output may be truncated copying 20 bytes from a string of length 491 [-Werror=stringop-truncation]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
   96 |                                   __glibc_objsize (__dest));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘int get_ata_identify(int, HDDInfo*)’ at /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/reader/hdd_reader/hdd_reader.cpp:230:10:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ output may be truncated copying 40 bytes from a string of length 457 [-Werror=stringop-truncation]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
   96 |                                   __glibc_objsize (__dest));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘int get_nvme_identify(int, HDDInfo*)’ at /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/reader/hdd_reader/hdd_reader.cpp:342:10:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ output may be truncated copying 20 bytes from a string of length 4091 [-Werror=stringop-truncation]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
   96 |                                   __glibc_objsize (__dest));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~
In function ‘char* strncpy(char*, const char*, size_t)’,
    inlined from ‘int get_nvme_identify(int, HDDInfo*)’ at /home/kenji/autoware/src/universe/autoware.universe/system/system_monitor/reader/hdd_reader/hdd_reader.cpp:348:10:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:95:34: error: ‘char* __builtin_strncpy(char*, const char*, long unsigned int)’ output may be truncated copying 40 bytes from a string of length 4071 [-Werror=stringop-truncation]
   95 |   return __builtin___strncpy_chk (__dest, __src, __len,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
   96 |                                   __glibc_objsize (__dest));
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~
```

### Purpose

To resolve warnings.

### Possible approaches

I'm not sure.

### Definition of done

No error after `-Wno-stringop-truncatio` is removed.

kenji-miyake, 2022-05-06:
@ito-san Could you please take a look at this? :pray:

ito-san, 2022-05-09:
@kenji-miyake Could you please let me know how to reproduce this?

kenji-miyake, 2022-05-09:
@ito-san Just removing this line.
https://github.com/autowarefoundation/autoware.universe/blob/391ebac4ebf8aeae386c2726929cdb1b334e0b7c/system/system_monitor/CMakeLists.txt#L8

It might not be reproduced with `Ubuntu 20.04`. I used `22.04`

ito-san, 2022-05-09:
I found the source code is not up-to-date. Thanks.

closed 2022-05-19 by kenji-miyake