autowarefoundation/autoware_universe #867: Add more detail about the min_behavior_stop_margin parameter to the Obstacle Stop Planner documentation

LalithVipulananthan, 2022-05-09:
In https://github.com/autowarefoundation/autoware/discussions/270, @08beeqtufail-sudo described a use case where a stop line is created (eg: at an intersection) but the ego vehicle stops earlier than expected due to an obstacle that exists just beyond the stop line. 

@takahoribe [explained that this behaviour can be managed with the `stop_planner.min_behavior_stop_margin` parameter](https://github.com/autowarefoundation/autoware/discussions/270#discussioncomment-2710929), and provided the following detail:

The stop_planner 's behavior is like this:

 - Set `v=0` at the point from `stop_margin` distance from an obstacle.
 - If another stop point already exists between an obstacle and ego (like in this case), set `v=0` at the point from `min_behavior_stop_margin` distance from an obstacle.

Generally `min_behavior_stop_margin` is set to be smaller than the `stop_margin`. See the [default parameter](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/obstacle_stop_planner/config/obstacle_stop_planner.param.yaml/#L4-L5). 

If you tune the parameter `min_behavior_stop_margin`, the vehicle will move as you expect.

The information above should be added to the [existing Obstacle Stop Planner documentation](https://autowarefoundation.github.io/autoware.universe/main/planning/obstacle_stop_planner/#obstacle-stop-planner_1).

satoshi-ota, 2022-05-16:
@LalithVipulananthan @TakaHoribe
May I close this issue since more detail documentation about the `min_behavior_stop_margin` has been added in https://github.com/autowarefoundation/autoware.universe/pull/880 ?

TakaHoribe, 2022-05-20:
Addressed in https://github.com/autowarefoundation/autoware.universe/pull/880

closed 2022-05-20 by TakaHoribe