autowarefoundation/autoware_universe #868: System_monitor build failure on Nvidia Xavier platform

Sharrrrk, 2022-05-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

system_monitor reported error building on Xavier.

```
--- stderr: system_monitor                                                                                                                                                   
/home/nv/ws/autoware/src/universe/autoware.universe/system/system_monitor/src/gpu_monitor/tegra_gpu_monitor.cpp: In member function ‘virtual void GPUMonitor::checkThrottling(diagnostic_updater::DiagnosticStatusWrapper&)’:
/home/nv/ws/autoware/src/universe/autoware.universe/system/system_monitor/src/gpu_monitor/tegra_gpu_monitor.cpp:130:80: error: unused parameter ‘stat’ [-Werror=unused-parameter]
  130 | void GPUMonitor::checkThrottling(diagnostic_updater::DiagnosticStatusWrapper & stat)
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/gpu_monitor_lib.dir/build.make:76: CMakeFiles/gpu_monitor_lib.dir/src/gpu_monitor/tegra_gpu_monitor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:602: CMakeFiles/gpu_monitor_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< system_monitor [9min 1s, exited with code 2]
```

### Expected behavior

Build success

### Actual behavior

Build error

### Steps to reproduce

Build

### Versions

- OS: Ubuntu 20.04
- ROS2: Galactic
- Autoware: main with Docker

### Possible causes

Unused variable

kenji-miyake, 2022-05-09:
Related? #270 (But seems another problem.)
@ito-san cc @KeisukeShima Could you comment on this? :pray:

ito-san, 2022-05-09:
> Related? #270 (But seems another problem.) @ito-san cc @KeisukeShima Could you comment on this? pray

@kenji-miyake Yes, checking thermal throttling is not implemented yet, so we are getting unused parameter error.

Sharrrrk, 2022-05-09:
> 

I think I could add a maybe_unused tag to the variable temporarily.

kenji-miyake, 2022-05-09:
@Sharrrrk Looks good. :+1: 
But please note that there will be other issues like #270.

PR #869 (2022-05-09): fix(system_monitor): fix build error on tegra platform

## Description

Resolve #868 

## Related links

#868 

## Tests performed

Nvidia AGX Xavier

## Notes for reviewers

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [ ] The PR follows the [pull request guidelines].
- [ ] The PR has been properly teste [... cut]
closed 2022-05-09 by Sharrrrk