autowarefoundation/autoware_universe #874: Poor dummy pointcloud on planning simulator

yukkysaito, 2022-05-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

Pointcloud should have some vertical layers, but, in the latest version, it has only 1 layer.

### Expected behavior

https://github.com/autowarefoundation/autoware.universe/pull/527#pullrequestreview-955895269

### Actual behavior

Pointcloud should have some vertical layers, but, in the latest version, it has only 1 layer.

### Steps to reproduce

1. launch planning simulator on following tutorials.
2. add dummy car or pedestrian

yukkysaito, 2022-05-09:
cc @HiroIshida

mitsudome-r, 2022-05-10:
@yukkysaito Is anyone from TIER IV going to work on this? If so, please assign this issue to the person.

HiroIshida, 2022-05-10:
@yukkysaito 
As for vertical layers, I will check and fix this if the issue is caused by https://github.com/autowarefoundation/autoware.universe/pull/527#pullrequestreview-955895269

@taikitanaka3 
```
dummy object's shape is also strange too.
```
Is that caused because the pointcloud is generated one side of the car, instead of all around the car? If so, it is intended behavior of my past PR.

closed 2022-05-11 by yukkysaito