autowarefoundation/autoware_universe #888: The lidar_centerpoint is now built with stderr

h-ohta, 2022-05-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=ON --catkin-skip-building-tests --parallel-workers 8  --packages-up-to lidar_centerpoint
Starting >>> autoware_auto_geometry_msgs
Starting >>> autoware_auto_mapping_msgs
Starting >>> autoware_lint_common
Starting >>> tier4_debug_msgs
Finished <<< autoware_lint_common [0.46s]                                                                                                                                                   
Starting >>> autoware_cmake
Finished <<< autoware_cmake [0.47s]                                                                                                                                                   
Finished <<< autoware_auto_geometry_msgs [9.07s]                                                                                                                   
Starting >>> autoware_auto_perception_msgs
Finished <<< autoware_auto_mapping_msgs [9.74s]                                                                                                                
Starting >>> autoware_auto_planning_msgs
Finished <<< tier4_debug_msgs [12.4s]                                                                                                                              
Finished <<< autoware_auto_planning_msgs [22.9s]                                                                                      
Starting >>> tier4_autoware_utils
Finished <<< autoware_auto_perception_msgs [25.4s]                                                                    
Finished <<< tier4_autoware_utils [17.4s]                       
Starting >>> lidar_centerpoint
[Processing: lidar_centerpoint]                                  
--- stderr: lidar_centerpoint                                     

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com//uc?id=1KFhmA4oFT6CtZx5806QeMzn5H2tKa3oD 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com//uc?id=1iyk5VoQ4uNBGPZwypVZIMjSuSYAI1RxP 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com//uc?id=13aYPRHx17Ge4BqxzW9drAUSWTppjtUV5 

 	Cannot retrieve the public link of the file. You may need to change
	the permission to 'Anyone with the link', or have had many accesses. 

You may still be able to access the file from the browser:

	 https://drive.google.com//uc?id=14PJ_L3Jpz6Yi8GzoctVOEbGWcaCLArGp 
...
---
Finished <<< lidar_centerpoint [43.2s]

Summary: 9 packages finished [1min 34s]
  1 package had stderr output: lidar_centerpoint
```

This problem seems to occur neceessary models are not downloaded.

### Expected behavior

The lidar_centerpoint is built with no stderr.

### Actual behavior

The lidar_centerpoint is now built with stderr.

### Steps to reproduce

```
$ git clone git@github.com:autowarefoundation/autoware.git
$ cd autoware
$ mkdir src
$ vcs import src < autoware.repos
$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=ON --catkin-skip-building-tests --parallel-workers 8 --packages-up-to lidar_centerpoint
```

### Versions

- Ubuntu20.04
- ROS2 galactic
- autoware.universe: https://github.com/autowarefoundation/autoware.universe/tree/bb96ae7b56085a6054ad1ade970bad399faef35a

kenji-miyake, 2022-05-12:
@h-ohta cc @Shin-kyoto Probably the same reason.
https://github.com/autowarefoundation/autoware.universe/pull/871

h-ohta, 2022-05-12:
@kenji-miyake Thank you for your information. I tried it, and I found that problems looks like resolved.
I'll make PR to fix this issue.

PR #889 (2022-05-12): fix(lidar_centerpoint): fix google drive url to avoid 404
## Description

- resolves #888

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before merging.

- [ ] Th [... cut]
closed 2022-05-12 by h-ohta