autowarefoundation/autoware_universe #918: Tensorrt_yolo issue in autoware universe

xmfcx, 2022-05-17:


kaancolak, 2022-05-22:
@amtommy I tested YoloV3, YoloV4, and YoloV5l with the bag file provided by you. It looks like working well without any change current codebase. 

Could you share your steps for reproducing your detection results?

I shared result videos:

YOLO V5L:
[image]

YOLO V4:
[image]

YOLO V3:
[image]

amtommy, 2022-05-23:
@kaancolak  Thanks for your verification.
I compare my code with the current codebase.
I find the problem is the module uses the default anchor size[[link](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/tensorrt_yolo/src/nodelet.cpp#L57)], so I got the incorrect result.
Now I use the correct anchor size [[link](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/tensorrt_yolo/config/yolov3.param.yaml#L4)], and I get the same result as yours.
The problem is solved, thanks for your help.

closed 2022-05-23 by mitsudome-r