autowarefoundation/autoware_universe #949: Monitor processing time for huge package

taikitanaka3, 2022-05-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Accuracy and processing time are in the balance and it's necessary which module is heavy.

### Purpose

to monitor topic hz easily and more precisely without subscribing heavy topic
[image]

### Possible approaches

- run autoware with this PR and see in plot_juggler
- `ros2 topic hz ~` topic at every topic for each manually and remember the trend of topic hz change.
  - be careful that ros2 topic hz also needs subscription of topic so itself also needs computation cost and if you see heavy topic, for example hz of point cloud,  you need to turn off for every time.

### Definition of done

ros2 run plotjuggler plotjuggler and see `*/debug/processing_time_ms`

BonoloAWF, 2022-05-23:
@taikitanaka3 is your plan to integrate this with the node monitoring? Is this related to https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/282 and https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1233

taikitanaka3, 2022-05-24:
@BonoloAWF 
Thank you for information, but this issue is for planing and sensing performance monitoring so what you mentioned are not related.

taikitanaka3, 2022-05-30:
close this issue resolved by https://github.com/autowarefoundation/autoware.universe/pull/946

closed 2022-05-30 by taikitanaka3