autowarefoundation/autoware_universe #962: behavior_velocity_planner and behavior_path_planner cannot run in the same composable node container on Nvidia AGX Xavier

xqe2011, 2022-05-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I follwed the [documentation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/) for planning simulation. The node `/planning/scenario_planning/lane_driving/behavior_planning/behavior_velocity_planner` cannot start, but in same container, `/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner` can. So I edited `./launch/tier4_planning_launch/launch/scenario_planning/lane_driving/behavior_planning/behavior_planning.launch.py` to run them in two different containers. Everything works.
If you change the sequence of them in `behavior_planning.launch.py `, e.g `[behavior_path_planner_component, behavior_velocity_planner_component]` to `[behavior_velocity_planner_component ,behavior_path_planner_component]`, it can start. But when you launch the full `planning_simulator.launch.xml`, you will see `component_container` exited by code -11.
If I use `ros2 component load` to load manually, the command will hang.

### Expected behavior

Run `ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/Downloads/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit` then node `behavior_velocity_planner ` and `behavior_path_planner ` can start in same container.

### Actual behavior

Node `behavior_velocity_planner ` cannot start.

### Steps to reproduce

```
ros2 launch "$HOME/autoware/src/universe/autoware.universe/launch/tier4_planning_launch/launch/planning.launch.xml" vehicle_info_param_file:="$HOME/autoware/src/vehicle/sample_vehicle_launch/sample_vehicle_description/config/vehicle_info.param.yaml"
```
Then run `ros2 node list`, you will see
```
/launch_ros_6869
/planning/mission_planning/goal_pose_visualizer
/planning/mission_planning/mission_planner
/planning/mission_planning/mission_planning_container
/planning/mission_planning/transform_listener_impl_aaaade6d2ca0
/planning/planning_diagnostics/planning_error_monitor
/planning/scenario_planning/external_velocity_limit_selector
/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner
/planning/scenario_planning/lane_driving/behavior_planning/behavior_planning_container
/planning/scenario_planning/lane_driving/behavior_planning/transform_listener_impl_ffff800a8970
/planning/scenario_planning/lane_driving/motion_planning/motion_planning_container
/planning/scenario_planning/lane_driving/motion_planning/obstacle_avoidance_planner
/planning/scenario_planning/lane_driving/motion_planning/transform_listener_impl_ffff5c07c430
/planning/scenario_planning/motion_velocity_smoother
/planning/scenario_planning/parking/costmap_generator
/planning/scenario_planning/parking/parking_container
/planning/scenario_planning/parking/transform_listener_impl_aaaab52a26d8
/planning/scenario_planning/scenario_selector
/planning/scenario_planning/transform_listener_impl_aaaab49dece0
/planning/scenario_planning/transform_listener_impl_aaaab8b0c3a8
```
Node `behavior_velocity_planner` cannot start.

### Versions

```
nvidia@nvidia-desktop: uname -a
Linux nvidia-desktop 5.10.65-tegra #1 SMP PREEMPT Mon May 16 20:58:07 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux
```
Autoware version: commit 9c3047e0
Code built from source by following the documentation, because tensorRT deprecated `setMaxWorkspaceSize`, so I add a compiler flags `-Wno-error=deprecated-declarations` to CMakeLists.txt to make the compilation success.

takayuki5168, 2022-09-22:
@xqe2011 Hi, thanks for your report, and sorry to be late to reply.

I don't have any idea to fix this issue in a proper way. If you still struggle with this issue, I will have a discussion internally at TIER IV.

takayuki5168, 2022-11-29:
@xqe2011 Please reopen this issue if the issue still occurrs.

closed 2022-11-29 by takayuki5168