autowarefoundation/autoware_universe #966: Yaw calculation is unstable in obstacle_avoidance_planner when skipping optimization.

takayuki5168, 2022-05-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

We can skip optimization in obstacle_avoidance_planner (then, just resample path from behavior module).
In this condition, yaw calculation is unstable and this causes some failure in other modules (e.g. detection area in obstacle stop planner is expanded, and it detects obstacles outside the trajectory, then do stop planning).

### Expected behavior

Yaw calculation is stable

### Actual behavior

Yaw calculation is unstable

### Steps to reproduce

1. fix yaw calculation algorithm

kosuke55, 2022-05-26:
skip_optimization=false
[image]
skip_optimization=true
[image]

takayuki5168, 2022-05-26:
thanks
cc @tkimura4

mitsudome-r, 2022-05-31:
@takayuki5168 Do you already have an idea how to fix the issue? If you are planning to work on this tasks, please assign yourself as assignee. Otherwise, please attach help-wanted tag so that we can assign someone else to look into this.

takayuki5168, 2022-06-01:
@mitsudome-r I don't look into this issue, so I have no idea so far. I attached help-wanted.
Let me say again that this issue will not occur with the default setting (skip_optimization = false in obstacle_avoidance_planner)

mitsudome-r, 2022-06-07:
I have assigned @mehmetdogru so that he can investigate when he has time. Any other help is appreciated.

xmfcx, 2022-06-14:
@klu17 will look into this.

mitsudome-r, 2022-06-17:
@takayuki5168 Could you provide the steps to reproduce the error (including sharing maps, scenarios that you used to find the error)

kosuke55, 2022-06-17:
@mitsudome-r @takayuki5168 
Any map(like [sample](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/#steps)) can be used, just set skip_optimization to true(use rqt_reconfigure or edit launch file) and draw a path with curves. 
https://user-images.githubusercontent.com/39142679/174250077-60972945-280d-4340-a043-5257bd48b232.mp4

mitsudome-r, 2022-06-29:
Steps to reproduce:
* Follow [installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/#installing-autoware) instruction to install Autoware
* Follow [planning simulator](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#planning-simulation) instructions to download the sample map and launch planning simulator.
* Set current pose and goal pose as shown in the image.
[image]
* In rviz display, turn on the debug marker in Planning->ScenarioPlanning->LaneDriving->MotionPlanning->DebugMarker. This will visualize the footprint of the vehicle on the trajectory
[image]

* start rqt_reconfigure 
```
source autoware/install/setup.bash
ros2 run rqt_reconfigure rqt_reconfigure
```
* Type obstacle_avoidance in the filter key and expand to go to the configuration window for the `obstacle_avoidance_planner` node.
* tick/untick option.skip_optimization and confirm that footprint is not smooth when skip_optimization is on.
 
[image]

mitsudome-r, 2022-07-05:
@klu17 Do you have any updates on this issue?

klu17, 2022-07-06:
No, I don't. I'll unassign.

angry-crab, 2022-07-19:
It seems that when optimization is skipped, the output is simply the path message being received. `obstacle_avoidance_planner` does nothing to the planned path. I'll investigate further.

takayuki5168, 2022-07-19:
@angry-crab 
obstacle avoidance planner does resampling of x, y, and yaw to the path from the behavior module.
https://github.com/autowarefoundation/autoware.universe/blob/69c2a1815034fb53f8c6a1faa5546fb6b49fccff/planning/obstacle_avoidance_planner/src/node.cpp#L1400-L1401

The yaw in the path from the behavior module is not smoothed well. So just resampling cannot deal with smooth yaw interpolation.
You can see how the yaw in the path from the behavior is not smoothed with PathFootprint rviz plugin.

angry-crab, 2022-07-21:
> @angry-crab obstacle avoidance planner does resampling of x, y, and yaw to the path from the behavior module. [...]

Thank you for the information. If so, isn't the unstable output expected somehow? Since this only happens when optimization is skipped, what would be our expectation for resampling methods?

angry-crab, 2022-07-21:
[image]

[image]

takayuki5168, 2022-07-21:
@angry-crab 
This unstable yaw in the behavior module(?) is not expected. We don't know why this unstable yaw happens. could you look into this issue more. Is this information enough for you to start dealing with this issue?

angry-crab, 2022-07-25:
@takayuki5168 Thank you for the information. Further investigation indicates that the issue is not related to `obstacle_avoidan
ce_planner`. The output trajectory of `obstacle_avoidance_planner` is as follows:
[image]

The plot obtained as @mitsudome-r described is the output of `obstacle_stop_planner`. I'll investigate further. 

> Steps to reproduce:
>
> * Follow [installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/#installing-autoware) instruction to install Autoware
> * Follow [planning [...]

angry-crab, 2022-07-25:
FYI, interpolation does not seems good as well. 

[image]

Is this expected? @takayuki5168

takayuki5168, 2022-07-25:
@angry-crab Not expected.
obstacle_stop_planner is, as you know, after obstacle_avoidance_planner.
When skip_optimization=true, obstacle_avoidance_planner's output is unstable. Therefore obstacle_stop_planner's output is unstable as well. So the problem is obstacle_avoidance_planner (which does fine resampling) or behavior_velocity_planner (which does not do find resampling).

angry-crab, 2022-07-26:
@takayuki5168 I think the issue is caused by interpolation. I plotted raw interpolated results against original input. 

XY:
[image]

Yaw:
[image]

X and Y look good, but yaw interpolation is problematic. Is there any specific reason that the interpolation was done for each dimension separately? And do you have any suggestion on how to fix this?

takayuki5168, 2022-07-26:
@angry-crab Nice plot.
I feel that the problem is yaw of the behavior velocity planner rather than spline interpolation of obstacle avoidance planner.
For example, this yaw of behavior velocity planner (red points plot) is not smooth at all.
Even if we do linear interpolation for yaw in obstacle avoidance planner, the yaw will change a bit suddenly.
[image]

Could you look into why yaw from behavior velocity planner is not so smoothed?

takayuki5168, 2022-07-27:
@angry-crab 
I heard that behavior velocity resamples path points with a certain interval and also makes sure that it samples the input path points (path with lane id from behavior_path_planner).

It seems this third red point in the plot is the sampled input path point. This implementation may be wrong
[image]

angry-crab, 2022-07-27:
@takayuki5168 Are you suggesting that the implementation of `behavior velocity planner` is wrong because it fails to output a required sampled path, or the implementation of `obstacle_avoidance_planner` is problematic because it could not detect this issue?

takayuki5168, 2022-07-27:
@angry-crab 

> Are you suggesting that the implementation of behavior velocity planner is wrong

Yes. This is because the behavior velocity planner's output path is not so smooth (Yaw angles change suddenly shown in the plot, which is not expected).

>  or the implementation of obstacle_avoidance_planner is problematic because it could not detect this issue?

For now, no. Because now we do not have a validation mechanism (to detect this issue) in the planning module.

angry-crab, 2022-07-29:
Further investigatoin indicates that such behavior is caused by interpolation. 
[image]

When there are two pose really close to each other in Euclidean distance but a somehow large fitting error would lead to such jump in yaw. One example is shown in the following image. 

[image]

Moreover, unlike `obstacle_avoidance_planner`, `behavior_velocity_planner` interpolates x,y,z, and velocity. Yaw is calculated by `(y[i+1]-y[i])/(x[i+1]-x[i])` 
https://github.com/autowarefoundation/autoware.universe/blob/5571338fd71974b31a51966eb1f8d14bafc48d2e/planning/behavior_velocity_planner/src/utilization/path_utilization.cpp#L236-L240
This is vulnerable to interpolation error and numeric accuracy. Even though the interpolation of x and y look good in picture, tiny horizontal or vertical error would be amplified during calculation when two consecutive poses are too close.  

Some possible ways to fix it:
1. increasing `epsilon` to 0.5 seems to solve the issue. (quick and dirty)
2. linear interpolation. (not differentiable near joints, but should be able to remove the unstable changes)
3. get yaw from vehicle / localization (not sure if it helps or not)
4. 2d pose interpolation with relaxing constraints (does not have to pass the joints)

takayuki5168, 2022-08-01:
@angry-crab 
Thank you so much for your further investigation.

Could you look into the reason why the yaw of the third red point has an error? I guess if this third red point is removed, the spline interpolation will be stable.
[image]

As you know, yaw in the behavior path planner is calculated by x and y. So the reason may be these x and y are not well-smoothed?

angry-crab, 2022-08-02:
@takayuki5168 
The plotted XY and yaw from `behavior_velocity_planner` is as follows:

XY

[image]

Yaw(Ignore the last point, it was added to make sure the size of output matches x and y)

[image]

From these graphs, we can see that the third yaw is generated by the third and fourth point of interpolated x and y. 
It is true that this part is a little bit concave compared to points in the neighborhood. Again, I believe it is because of interpolation and this kind of error cannot be avoided with interpolation methods. You can see that the input is quite discrete and this gives rise to the interpolated error. I'm not sure how "smooth" we are expecting from interpolation. But the results indicate that it is difficult to expect too much from it.

angry-crab, 2022-08-04:
After discussing with @takayuki5168 , we believe there is something wrong with the implementation of interpolation because the input points should always overlap with the output points since the original points are kept.

angry-crab, 2022-08-05:
@takayuki5168 I think the previous error of interpolation we discussed during the meeting is actually caused by plotting.
[error](https://user-images.githubusercontent.com/93699235/182310194-953d98f2-cc1d-42d9-a133-3f8ffebaf5ee.png)
I plotted the points separately and found out that the inputs do exist in the interpolated outputs. The input overlaps with the output as expected. The reason it appears red in the image is because I plotted the input after output.

takayuki5168, 2022-08-08:
@angry-crab Thank you for your report.
I think now I got the cause.

As shown in the figure, there is a slight difference of s (s is the length along the path) between behavior path points (black ones with a wide interval, input points for spline interpolation) and behavior vel points (red ones, the points with certain interval) since `black l1` is not `red l1 + l2 + l3`.
black a2 and red a4 are the same points, but in the s-x graph, the s is different.
[image]

angry-crab, 2022-08-15:
@takayuki5168 Thank you for the graph. Sorry for this late reply because I was off last week. I think the issue is caused by limitation of 1d interpolation. The following plot is y vs s. You could see this is not a "function" because each input s might correspond to two different y. 

[image]

This creates ambiguity when we are looking for intervals to be used for interpolation.

angry-crab, 2022-08-16:
@takayuki5168 Thank you for the graph. It is truth that the black l1 is shorter than the red segments added up. However, it would not have too much impact on the result because even though a2 and a4 differs, they could just be considered just as different points. Further, the `has_almost_same_value` check is going to add more points anyway. 
Here is another plot. I tried to isolate the interpolation part and created a denser input set to visualize the shape of interpolation function. Red points are generated with a step size of 0.1. 
[image]

1. I think the interpolation itself is good. I also tried to create artificial inputs with random noises to see its fault tolerance. If the random noise is within 0.1, the output looks good. 
2. I feel it is difficult to go further with removing certain points.
3. Back to our original problem, I guess there are two ways that I can think of to solve the issue. (1. keep a certain distance between consecutive poses. 2. Don't interpolate yaw, calculate it instead with consecutive poses in `behavior_avoidance_planner`. )

Please let me know your thoughts on this.

takayuki5168, 2022-08-18:
@angry-crab 
Seeing this plot, I remembered that matplotlib discretized the large values when plotting. We can see that some successive red points have the same value, and that's what I said.
[image]

In this plot, the same matplotlib issue may occur.
I mean, the second red point and the second green point seem to have to same value which we considered as the problem, but 
 I guess these points does not have the same value.
[image]

takayuki5168, 2022-08-18:
However, as you pointed in this plot, there still is the issue of interpolation error.
[image]

angry-crab, 2022-08-19:
After discussing with @takayuki5168, I believe the issue is due to pairs of poses that are too close in distance, especially near the input poses of interpolation. Since interpolation forces the second derivative at input pose to be zero, close pairs of poses are assigned similar values. This fact give rise to a sudden change in yaw for the next consecutive poses.

angry-crab, 2022-08-23:
In conclusion, adjusting `epsilon` solves the issue. Setting `epsilon = 0.2`, we have the following trajectory. 
[image]

angry-crab, 2022-08-26:
#1687 
This PR fixes the issue somehow, but it is correcting the yaw afterwards.

PR #1687 (2022-09-05): fix(obstacle_avoidance_planner, interpolation): improve slerp and fix yaw

## Description

#1499 Fix yaw interpolation and improve performance.
Since the current yaw interpolation is not stable, I just switched to the implementation of `fillYawInTrajectoryPoint`.
This PR also somehow fixes #966 by correcting yaw afterwards.

## Before

```
[component_container_mt-15] [INFO] [1661409128.691758302] [obstacle_avoidance_planner.time]:     generateFineTrajectoryPoints:= 3.928 [ms]
[component_container_mt-15] [INFO] [1661409128.756615001] [obstacle_avoidance_planner.time]:     generateFineTrajectory [... cut]
closed 2022-09-05 by angry-crab