autowarefoundation/autoware_universe #977: concatenate data nodelet does not transform point cloud

kaancolak, 2022-05-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

 I tried to run some sensing/perception nodes standalone. When I test the concatenate_data nodelet, I decided there is some problem with transformPointCloud functions.  tf_ros::Buffer pointer objects declaring [here](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/include/pointcloud_preprocessor/concatenate_data/concatenate_data_nodelet.hpp#L139) and used [here](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/src/concatenate_data/concatenate_data_nodelet.cpp#L176) but never initialize. 

### Expected behavior

Transform and concatenate point clouds without any problem.

### Actual behavior

If the given point cloud contains a different "frame_id", nodelet crash with a segmentation fault.

### Steps to reproduce

1. Run concatenate_data nodelet using multiple point cloud that has different "frame_id"

VRichardJP, 2022-05-27:
I think this is solved by PR https://github.com/autowarefoundation/autoware.universe/pull/933

miursh, 2022-05-27:
Is https://github.com/autowarefoundation/autoware.universe/pull/933 ready to be merged?

shmpwk, 2022-05-27:
@kaancolak 
I merged #933, could you try it again?

kaancolak, 2022-05-27:
Thanks, it look working well. 
Resolved with #933