autowarefoundation/autoware_universe #980: Canot stop before blocking obstacle in parking

alanmengg, 2022-05-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When  simulating in parking lot , if an obstacle blocks the trajectory and there is no available trajectory to reach the goal, the car can't stop before the obstacle and passes through it.
[image]

### Expected behavior

Vehicle should stop if there is no available path. 

### Actual behavior

Vehicle can't stop and passes through the obstacle.

### Steps to reproduce

1. Run the planning simulator with parking lot, set a goal and engage.
2. Put a obstacle to block the ego vehicle.

### Versions

ubuntu 20.04
galactic
universe:main

### Possible causes

the published stop trajectory doesn't actually work because it's  current pose, and control module can't stop the car with 0 distance.

xmfcx, 2022-05-31:
@alanmengg could you please share the resources and commands to reproduce this issue? Like maps used.

alanmengg, 2022-06-01:
> here is the lanelet map I used for the simulation
[lishui_parking.zip](https://github.com/autowarefoundation/autoware.universe/files/8811260/lishui_parking.zip)

alanmengg, 2022-06-01:
> 
here is the lanelet map I used for the simulation
[lishui_parking.zip](https://github.com/autowarefoundation/autoware.universe/files/8811270/lishui_parking.zip)

mitsudome-r, 2022-06-07:
@TakaHoribe It would be nice if you can assign someone from TIER IV's planning team to take a look at this.

shmpwk, 2022-06-30:
Does this PR solve the issue?
https://github.com/autowarefoundation/autoware.universe/pull/985

raminterpl, 2022-08-08:
@alanmengg : Hi, can you pls share me map used for this parking scenario

alanmengg, 2022-08-12:
> @alanmengg : Hi, can you pls share me map used for this parking scenario

here is the map
[map.zip](https://github.com/autowarefoundation/autoware.universe/files/9313732/map.zip)
p

raminterpl, 2022-08-23:
@alanmengg : I confirm above mentioned issue. Ego vehicle is not stopping before blocking obstacle in parking
[image]
.

raminterpl, 2022-08-23:
@alanmengg : I confirm the same behavior . Ego vehicle is not stopping if there is an obstacle in trajectory path in Parkin
[image]
g

shmpwk, 2022-09-02:
@alanmengg 
Looks like this issue is not completed yet?

mitsudome-r, 2022-09-13:
@alanmengg friendly ping.

closed 2022-10-03 by shmpwk