autowarefoundation/autoware_universe #993: Vehicle stops ~1m further away than goal point or stop line

VRichardJP, 2022-05-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While testing autonomous driving in our vehicle, I observed that the car always stops ~1 meter after the goal point or stop line.

For example here, I set the target goal a few meters in front of the vehicle, then engage AD. The vehicle drives toward the goal, but instead of stopping exactly at the goal location (with base_link at the goal position) the vehicle drives too far and parks ~1 meter after:
[image]

I also observed the same issue with a stop line. The vehicle stop systematically crosses the stop line and parks ~1 meter after.

I have never encountered that issue with autoware.ai or architecture proposal (ros1). My vehicle and its configuration are the same than what I used to test architecture proposal.

### Expected behavior

The vehicle base_link should stop exactly at the goal position

### Actual behavior

The vehicle stops ~1m after goal

### Steps to reproduce

1. run autoware.launch in vehicle
2. set target goal
3. engage AD

### Versions

- Ubuntu 20.04
- ROS2 galactic
- autoware main branch

### Possible causes

- wrong vehicle configuration?
- delay on decision/cmd messages?

kenji-miyake, 2022-05-30:
Related?
https://github.com/orgs/autowarefoundation/discussions/376#discussioncomment-2845593

VRichardJP, 2022-05-30:
@kenji-miyake thanks for pointing out. It is maybe the same issue (except my vehicle brakes smoothly).
I will check `accel_brake_map_calibrator`

xmfcx, 2022-05-31:
@VRichardJP Are you using the planning simulator or a real vehicle? Could you please provide steps / (if possible) bag files to reproduce this issue?

VRichardJP, 2022-06-01:
@xmfcx I am using a real vehicle. I did not record any rosbag of the situation, but maybe it cannot be reproduced with the simulator.

TakaHoribe, 2022-06-05:
@VRichardJP I suggest to use the plutJuggler with trajetory_follower config. It will visualize useful information as an attached figure and we'll see the cause of the overshoot.

The configuration file is in `src/autoware/universe/control/trajectory_follower_nodes/config/plot_juggler_trajectory_follower.xml`.

[image]

JianKangEgon, 2022-06-06:
From my own road test experience,  if the ego vehicle velocity is relatively small and the distance between vehicle and goal is smaller than one pre-defined threshold parameter named "th_arrived_distance_m", then the vehicle will stop and autoware will be disenaged. 

Perhaps this parameter may help you.

TomoyukiaW, 2022-06-16:
@TakaHoribe Thank you for your advice. I'm working with Vincent. I captured the longitudinal controller's control profile. Looking at the velocity, target velocity got zero before the current velocity. Does this means the vehicle stopped slightly beyond the time (~= location) that the longitudinal controller targeted? If so, does the controller think the vehicle ahead the real position...?
[image]
[image]

TomoyukiaW, 2022-06-16:
@JianKangEgon Thank you for your comment. But I think th_arrived_distance_m is the margin the system think the vehicle arrived at the goal. I think this issue is that the system can't control the vehicle to stop the targeted point.

TomoyukiaW, 2022-06-16:
@TakaHoribe Thank you for your advice. I'm working with Vincent. I checked the target velocity and the current velocity of output of longitudial_controller. After the target velocity got to zero, the current velocity was still high. I thought the brake value from raw_vehicle_cmd_converter was not enough to the target acceleration (-0.3m/s^s) in stopping state. Then I modified the brake map values around 0.3, 1.39m/s manually, and vehicle base_link matched to the goal point. (My accel_brake_map_calibrator result didn't work well.) But I feel a little bit not smooth. If you have any advice to tune the brake_map for stopping behavior, could you let us know?

### Default brake_map
[image]
[image]

### Modified brake_map
[image]
[image]

xmfcx, 2022-07-19:
@TakaHoribe @TomoyukiaW @VRichardJP are there any updates on this?

VRichardJP, 2022-07-25:
After some calibration I think we solved the issue. Now our vehicle manages to consistently park at target location.

closed 2022-07-25 by VRichardJP