autowarefoundation/autoware_universe #1008: Test the perception stack on the raw lidar data

xmfcx, 2022-05-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Follow up from: https://github.com/autowarefoundation/autoware.universe/issues/562#issuecomment-1102662448

Bag and pcap files: https://autoware-files.s3.us-west-2.amazonaws.com/collected_data/2022-04-14_autocore-lidar-bag-pcap/Lidar_Data_220414_bag_pcap.zip

### Purpose

AutoCore.ai has recorded and shared lidar pcap and bag files from their autonomous vehicle. This task aims to test the lidar perception stack of Autoware Universe.

### Possible approaches

The bag files contain a single topic with:
- name: `/sensing/lidar/top/rectified/pointcloud`
- type: `sensor_msgs/msg/PointCloud2`

Run the perception stack on the data and obtain locally tracked objects.

### Definition of done

- [ ] The required nodes determined and run.
- [ ] Ground segmentation works.
- [ ] The object bounding boxes are obtained.
- [ ] Local tracking results are obtained. (Without localization)
- [ ] List of nodes that were run are shared under this issue.
- [ ] Video(s) of the RViz showing the end result uploaded to youtube.

kaancolak, 2022-06-10:
I tested the perception pipeline with a given Velodyne point cloud bag file. I used velodyne_top as a base frame for each node.  multi_object_tracker contains a hard-coded parameter as a "base_link"   [here](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/multi_object_tracker/src/multi_object_tracker_core.cpp#L222) for this reason running a multi-object tracker without "base_link" need an external change in the package.

List of nodes that were run:
-  common_ground_filter
-  euclidean_cluster
-  detection_by_tracker_node
-  initial_cluster_shape_estimation
-  lidar_apollo_instance_segmentation or lidar_centerpoint
-  obstacle_pointcloud_based_validator_node
-  object_association_merger
- multi_object_tracker

Ground remover on trucs bag file : 
[image]

Ground remover on ped_vehicle bag file : 
[image]

Tracking result on trucs bag file with Apollo 3D detections : 
[image]

Tracking result on ped_vehicle bag file with Apollo  3D detections: 
[image]

Additionally, lidar_centerpoint returned too many false positive detected objects in the provided bag files. I increased the score_threshold up to 0.60 false-negative detections decreased but still continue. I used a local frame "velodyne_top" as a densification_world_frame_id. 

[image]

FYI:  @xmfcx @mitsudome-r @Sharrrrk

xmfcx, 2022-06-10:
Wow thanks for the tests @kaancolak!

The most of the detection markers are coming from behind in the RViz2 visualizations.
Do you know why is that the case?

I will open up issues to look into the problems if it is not the expected state.

Also the pedestrian detection video looks poor in terms of detecting and tracking the person: https://youtu.be/J8J2e_YI-SQ?t=7

@yukkysaito are these results expected or do you think these results can be improved by simple tuning of the parameters?

kaancolak, 2022-06-12:
I don't know the exact cause of this problem. I did not experience this problem when working with the "map" frame.

mitsudome-r, 2022-06-13:
The false-negative object might be caused by training set coming from a different lidar model. 

@yukkysaito or @aohsato Do you know how the current model was trained (e.g., sensor setup)? Also, is it possible to share the training script for the models?

aohsato, 2022-06-14:
Thank you for sharing! It's interesting.

Did you obtain your pcap data with Ouster OS1-64, correct?
About your result of centerpoint, in [this video](https://youtu.be/J8J2e_YI-SQ), there may be a sensor model gap with the training model, but I think this pointcloud density is enough to detect the pedestrian.

We have found a similar issue and @yukke42 is currently working on it. This problem is that the sampling at centerpoint is not uniform and depends on the order of pointcloud.

mitsudome-r, 2022-06-21:
@aohsato 
> We have found a similar issue and @yukke42 is currently working on it. This problem is that the sampling at centerpoint is not uniform and depends on the order of pointcloud.

Do you have any ETA on the fix? Please let us know if you would like help from the others.

yukke42, 2022-06-28:
Sorry for the late reply. I guess there are some points for worse performance:
- lidar_centerpoint requires tf between base_link (or velodyne_top) and map to stack multi-frame pointcloud. You can set 0 [here](https://github.com/autowarefoundation/autoware.universe/blob/458a32d0e132a01a092e3e3814176e564021e91c/perception/lidar_centerpoint/launch/lidar_centerpoint.launch.xml#L14) but it gets worse performance.
- The pointcloud data for model training is transformed to base_link of autoware.

We found that we can apply [voxel grid downsample filter](https://github.com/autowarefoundation/autoware.universe/blob/458a32d0e132a01a092e3e3814176e564021e91c/perception/lidar_centerpoint/launch/lidar_centerpoint.launch.xml#L14) to input cloud to detect close pedestrians with less overhead and less worse effect on the detection performance of other objects.

xmfcx, 2022-06-28:
For single shot detection, if the target is moving, setting densification parameter larger than 0 would create a trail. Is this a good way of doing this? @yukke42

yukke42, 2022-06-29:
> For single shot detection, if the target is moving, setting densification parameter larger than 0 would create a trail. Is this a good way of doing this? @yukke42

@xmfcx Yes. Densification makes better performance than only the current frame especially for pedestrian and bicycle. This method is generally used in recent 3D object detection papers.

mitsudome-r, 2022-07-05:
@kaancolak Could you create a separate issue for current issues that you see in perception stack?

mitsudome-r, 2022-08-02:
From the ASWG meeting, we have decided to close this issue and let LeoDrive/ITRI to create more specific issues once they find any problems during integration.
@mehmetdogru will have a final look on this issue to see if there are still remaining issues from the past discussion.

mehmetdogru, 2022-08-02:
> [I will open up issues to look into the problems if it is not the expected state.](https://github.com/autowarefoundation/autoware.universe/issues/1008#issuecomment-1152378043)

@xmfcx please open up the issues if you already didn't do so.

I think we can close this issue.

closed 2022-08-02 by mehmetdogru