autowarefoundation/autoware_universe #1009: Sometimes the sensing module doesn't publish concatenated pointcloud

yukkysaito, 2022-05-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

sometimes the sensing module cannot launch concatenated filter with the following error
```
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'concatenate_data' of type 'pointcloud_preprocessor::PointCloudConcatenateDataSynchronizerComponent' in container 'pointcloud_container': Failed to find class with the requested plugin name.
```

### Expected behavior

launch concatenated filter without error

### Actual behavior

launch concatenated filter without error

### Steps to reproduce

play rosbag replay simulator in tutorials

wep21, 2022-05-31:
Could you confirm that the requested plugins above are in the list which is shown with `ros2 component types`?

yukkysaito, 2022-06-01:
> ros2 component types
@wep21 This is the result of `ros2 component types`
[image]

miursh, 2022-06-02:
I tried launching the sensing module 10 times each on 2 PCs, but have not been able to reproduce it here

mitsudome-r, 2022-06-07:
We will wait for further reports if any other developers can reproduce the error.
@yukkysaito If no one can reproduce the error, we might have to close this issue since we have no way of doing further investigation.

yukkysaito, 2022-06-10:
@esteve I launch it following the tutorial
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/
In TIER IV, some people meet this error.

mitsudome-r, 2022-06-21:
@yukkysaito ping

VRichardJP, 2022-06-27:
Hi,

I recently started to have this very error using `autoware_launch.launch.xml`. It does not occur every single time, but when it does it can be very frustrating.

Although the message seems to point out there is an issue with the plugin, it seems to me the problem appears when the plugin is tried to be loaded before the container is actually started.

The way I have found to fix that problem is to move the launch of the container from `autoware.launch.xml` to a separate launch file, which is to be started before `autoware.launch.xml` (same for other launch files). I have been using this for a few days and have not encountered the issue anymore since.

Like this `pointcloud_container.launch.xml`:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <!-- Pointcloud container -->
  <arg name="pointcloud_container_name" default="pointcloud_container"/>
  <group>
    <include file="$(find-pkg-share autoware_launch)/launch/pointcloud_container.launch.py" if="$(var use_pointcloud_container)">
      <arg name="use_multithread" value="true"/>
      <arg name="container_name" value="$(var pointcloud_container_name)"/>
    </include>
  </group>
</launch>
```

xmfcx, 2022-07-19:
We should further investigate why this happens in https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/pointcloud_preprocessor.launch.py

Also we need to test and see if it happens in humble too.

miursh, 2022-07-25:
The reason why this happens is because sensing.launch and perception.launch could start before the pointcloud container start, I think. 
In autoware_launch.launch.xml/logging_simulator.launch.xml,  pointcloud container must be launched first, but I suspect current launcher does not guarantee launch order.

yukkysaito, 2022-07-26:
>@yukkysaito When this error happens, if you restart the computer, does it start working again?
yes
sorry for late reply. 🙏

wep21, 2022-07-27:
> The reason why this happens is because sensing.launch and perception.launch could start before the pointcloud container start, I think.

`LoadComposableNodes` can load nodes even if the target container is launched after it, so I think the order of launch has nothing to do with this issue.

wep21, 2022-07-27:
Here is some test snippet.
```
from launch import LaunchDescription
from launch_ros.actions import LoadComposableNodes, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    container = Node(
        name='image_container',
        package='rclcpp_components',
        executable='component_container',
        output='both',
    )

    load_composable_nodes = LoadComposableNodes(
        target_container='image_container',
        composable_node_descriptions=[
            ComposableNode(
                 package='image_tools',
                plugin='image_tools::Cam2Image',
                name='cam2image',
                remappings=[('/image', '/burgerimage')],
                parameters=[{'width': 320, 'height': 240, 'burger_mode': True, 'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True}],
            ),
            ComposableNode(
                package='image_tools',
                plugin='image_tools::ShowImage',
                name='showimage',
                remappings=[('/image', '/burgerimage')],
                parameters=[{'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True}]
            ),
        ],
    )

    return LaunchDescription([load_composable_nodes])
```

```
from launch import LaunchDescription
from launch_ros.actions import LoadComposableNodes, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    container = Node(
        name='image_container',
        package='rclcpp_components',
        executable='component_container',
        output='both',
    )

    load_composable_nodes = LoadComposableNodes(
        target_container='image_container',
        composable_node_descriptions=[
            ComposableNode(
                 package='image_tools',
                plugin='image_tools::Cam2Image',
                name='cam2image',
                remappings=[('/image', '/burgerimage')],
                parameters=[{'width': 320, 'height': 240, 'burger_mode': True, 'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True}],
            ),
            ComposableNode(
                package='image_tools',
                plugin='image_tools::ShowImage',
                name='showimage',
                remappings=[('/image', '/burgerimage')],
                parameters=[{'history': 'keep_last'}],
                extra_arguments=[{'use_intra_process_comms': True}]
            ),
        ],
    )

    return LaunchDescription([container])
```

xmfcx, 2022-07-27:
@wep21 were you able to reproduce this issue?

For me 80% of the times I run the logging simulator launch file, do you know why it might be happening like this then?

mitsudome-r, 2022-08-02:
@miursh @wep21 Did you make any progress on this issue?

wep21, 2022-08-02:
I can't reproduce the issue, but from the log my colleague shared, I guess some components which are supposed to exist aren't found as below.
```
[component_container_mt-1] [ERROR] [1658719110.290187926] [pointcloud_container]: Failed to find class with the requested plugin name 'pointcloud_preprocessor::CropBoxFilterComponent' in the loaded library
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'crop_box_filter' of type 'pointcloud_preprocessor::CropBoxFilterComponent' in container 'pointcloud_container': Failed to find class with the requested plugin name.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/perception/object_recognition/detection/clustering/euclidean_cluster' in container '/perception/object_recognition/detection/clustering/euclidean_cluster_container'
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'voxel_grid_downsample_filter' of type 'pointcloud_preprocessor::VoxelGridDownsampleFilterComponent' in container 'pointcloud_container': Failed to find class with the requested plugin name.
[component_container_mt-1] [ERROR] [1658719110.292822374] [pointcloud_container]: Failed to find class with the requested plugin name
```

miursh, 2022-08-02:
I made the environment to reproduce this and am debugging, but still can't figure it out.

mitsudome-r, 2022-08-16:
@miursh is off for a week and won't be able to work on this until next week.

mitsudome-r, 2022-08-30:
I'm looking for the root cause, but there might not be a quick fix.
As a current workaround, LeoDrive is launching the container launch and component launch separately.

miursh, 2022-08-30:
I'm sorry I couldn't find enough time to work for this.
If someone else can handle this, please help.

lchojnack, 2022-09-14:
We find a way how to reproduce the issue. On my PC with 8 cores / 16 threads CPU when I'm running the command
```
stress --cpu 4
```
and then autoware, the issue almost always appears. Without `stress` command the issue doesn't occur. I'm not sure how exactly the CPU load can affect launching autoware. I will continue to work on this issue.

lchojnack, 2022-09-28:
During debugging I noticed that launch sequence is not a root cause. For whatever reason, when the failure occurs when   `libcompare_map_segmentation.so ` is loaded before  `libpointcloud_preprocessor_filter.so`

```
[INFO] [component_container_mt-1]: process started with pid [1359167]
[component_container_mt-1] [INFO] [1664351258.345340026] [pointcloud_container]: Load Library: /home/lukasz/projects/autoware_official_main/autoware/install/compare_map_segmentation/lib/libcompare_map_segmentation.so
[component_container_mt-1] [INFO] [1664351266.708360676] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<compare_map_segmentation::CompareElevationMapFilterComponent>
[component_container_mt-1] [INFO] [1664351266.708518922] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<compare_map_segmentation::DistanceBasedCompareMapFilterComponent>
[component_container_mt-1] [INFO] [1664351266.708557455] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<compare_map_segmentation::VoxelBasedApproximateCompareMapFilterComponent>
[component_container_mt-1] [INFO] [1664351266.708591027] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<compare_map_segmentation::VoxelBasedCompareMapFilterComponent>
[component_container_mt-1] [INFO] [1664351266.708623150] [pointcloud_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<compare_map_segmentation::VoxelBasedCompareMapFilterComponent>
[component_container_mt-1] [INFO] [1664351267.132734306] [perception.object_recognition.detection.voxel_based_compare_map_filter]: Filter (as Component) successfully created with the following parameters:
[component_container_mt-1]  - approximate_sync : false
[component_container_mt-1]  - use_indices      : false
[component_container_mt-1]  - latched_indices  : false
[component_container_mt-1]  - max_queue_size   : 5

[component_container_mt-1] [INFO] [1664351267.385921052] [pointcloud_container]: Load Library: /home/lukasz/projects/autoware_official_main/autoware/install/pointcloud_preprocessor/lib/libpointcloud_preprocessor_filter.so
[component_container_mt-1] [ERROR] [1664351267.430029960] [pointcloud_container]: Failed to find class with the requested plugin name 'pointcloud_preprocessor::PointCloudConcatenateDataSynchronizerComponent' in the loaded library
[component_container_mt-1] [ERROR] [1664351267.436821745] [pointcloud_container]: Failed to find class with the requested plugin name 'pointcloud_preprocessor::CropBoxFilterComponent' in the loaded library

[component_container_mt-1] [INFO] [1664351267.774941134] [pointcloud_container]: Load Library: /home/lukasz/projects/autoware_official_main/autoware/install/ground_segmentation/lib/libground_segmentation.so
[component_container_mt-1] [INFO] [1664351267.826568706] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<ground_segmentation::RANSACGroundFilterComponent>
[component_container_mt-1] [INFO] [1664351267.826816182] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<ground_segmentation::RayGroundFilterComponent>
[component_container_mt-1] [INFO] [1664351267.826862295] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<ground_segmentation::ScanGroundFilterComponent>
[component_container_mt-1] [INFO] [1664351267.826898226] [pointcloud_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<ground_segmentation::ScanGroundFilterComponent>
[component_container_mt-1] [INFO] [1664351268.760633951] [perception.obstacle_segmentation.common_ground_filter]: Filter (as Component) successfully created with the following parameters:
[component_container_mt-1]  - approximate_sync : false
[component_container_mt-1]  - use_indices      : false
[component_container_mt-1]  - latched_indices  : false
[component_container_mt-1]  - max_queue_size   : 5
[component_container_mt-1] [INFO] [1664351269.408068814] [pointcloud_container]: Load Library: /home/lukasz/projects/autoware_official_main/autoware/install/occupancy_grid_map_outlier_filter/lib/liboccupancy_grid_map_outlier_filter.so
[component_container_mt-1] [INFO] [1664351269.419959183] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<occupancy_grid_map_outlier_filter::OccupancyGridMapOutlierFilterComponent>
[component_container_mt-1] [INFO] [1664351269.420117594] [pointcloud_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<occupancy_grid_map_outlier_filter::OccupancyGridMapOutlierFilterComponent>
[component_container_mt-1] [WARN] [1664351283.256087367] [perception.obstacle_segmentation.occupancy_grid_map_outlier_filter]: New subscription discovered on topic '/perception/obstacle_segmentation/point cloud, requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```

When `libpointcloud_preprocessor_filter.so` is loaded before  `libcompare_map_segmentation.so` then everything is ok.
```
[INFO] [component_container_mt-1]: process started with pid [1345975]

[component_container_mt-1] [INFO] [1664350962.225026369] [pointcloud_container]: Load Library: /home/lukasz/projects/autoware_official_main/autoware/install/pointcloud_preprocessor/lib/libpointcloud_preprocessor_filter.so
[... cut ...]  sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[component_container_mt-1] [WARN] [1664350984.264477043] [perception.obstacle_segmentation.occupancy_grid_map_outlier_filter]: New subscription discovered on topic '/perception/obstacle_segmentation/point cloud, requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```

The main question is what determines the order of loading .so modules? 
Unfortunately, I don't know the answer.

xmfcx, 2022-10-04:
@lchojnack can you check if this https://github.com/autowarefoundation/autoware.universe/pull/1982 solves this issue?

PR #1982 (2022-10-13): fix(pointcloud_preprocessor): separate targets for base class
## Description

closes #1009 

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [ ] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before merging.

- [ ]  [... cut]
closed 2022-10-13 by shmpwk