autowarefoundation/autoware_universe #1011: Update planning simulation

mertyavuz41, 2022-05-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

* The official planning simulation test of the autoware.universe needs to be update, the ego vehicle cannot turn corners. 

Here I have a video:
https://netorgft4229778-my.sharepoint.com/:v:/g/personal/mert_adastec_com/EWJx8PusEylBnYmDQ5z6QWsBqQHtAIZKzgUrkcTtIbC_5g?e=CWEZdt

### Purpose

The official tests must run as pledged in the tutorial page, so it must not has any problem.

### Possible approaches

To update lanelet of the environment or fine-tune to the control parameters may solve the issue.

### Definition of done

Planning simulation must perform mission that mentioned on the tutorial page.

takayuki5168, 2022-05-31:
@mertyavuz41
Did you try the latest version of autoware? I guess it can solve this issue.

kenji-miyake, 2022-05-31:
@takayuki5168 Could you link some related PRs?

mertyavuz41, 2022-05-31:
@takayuki5168, I have been updated my autoware version(master) yesterday, but master branch has  a few newest commit. I will update again and I will try to run planning simulation once more.

takayuki5168, 2022-05-31:
@mertyavuz41
Sorry, I was using a different map. Now I confirmed that the latest autoware cannot turn at the corner you pointed.

takayuki5168, 2022-06-01:
@mertyavuz41 
@yukkysaito will work on the issue of the stop line by drivable area, by changing a map or somehow. So please wait for it :pray:

yukkysaito, 2022-06-01:
I'm working to have this issue resolved as soon as possible.

TakaHoribe, 2022-06-05:
@mertyavuz41 The plugin size issue has already been fixed in the latest version by the PR https://github.com/autowarefoundation/autoware.universe/pull/587. Could you update your version and check it? You need to update not only the autoware.universe but also the autoware_launch for rviz config. Just update the whole autoware is ok.

bariskapaligoz, 2022-06-06:
> @mertyavuz41 The plugin size issue has already been fixed in the latest version by the PR #587. Could you update your version and check it? You need to update not only the autoware.universe but also the autoware_launch for rviz config. Just update the whole autoware is ok.

Yes, rviz configuration fixed after autoware update.

yukkysaito, 2022-06-06:
@mertyavuz41 
I update the map :+1: you can download the same download link as before.
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/

mitsudome-r, 2022-06-21:
@mertyavuz41 @yukkysaito Can we close this issue?