autowarefoundation/autoware_universe #998: Avoid unnecessary copy in `Filter::computePublish`

VRichardJP, 2022-05-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The `pointcloud_preprocessor::Filter::computePublish` function seems to make unnecessary copy of intermediate point cloud:

https://github.com/autowarefoundation/autoware.universe/blob/a0ef0e9f4d9d074aa65dc7a1a08742b4b1a01921/sensing/pointcloud_preprocessor/src/filter.cpp#L162-L169

The `output` pointcloud is created then filled in `filter()`. Right after the cloud data is copied in a new `cloud_tf` pointer and never used again.

It is possible to verify that data is copied by using the following snippet:
```C++
  auto cloud_tf = std::make_unique<PointCloud2>(output);  // set the output by default
  printf("%p %p\n", (void*)&output.data[0], (void*)&cloud_tf->data[0]); 
```

Which shows on my machine:
```
0x5654f4df0650 0x5654f4d7a2c0
0x5654f4df0650 0x5654f4e355d0
0x5654f4e00f80 0x5654f4d7a2c0
0x5654f4df1660 0x5654f4f346d0
...
```

### Purpose

Remove unnecessary copy of heavy data

### Possible approaches

The function can be rewritten like so:

```C++
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////
void pointcloud_preprocessor::Filter::computePublish(
  const PointCloud2ConstPtr & input, const IndicesPtr & indices)
{
  auto cloud_tf = std::make_unique<PointCloud2>();

  // Call the virtual method in the child
  filter(input, indices, *cloud_tf);

  // Check whether the user has given a different output TF frame
  if (!tf_output_frame_.empty() && cloud_tf->header.frame_id != tf_output_frame_) {
    RCLCPP_DEBUG(
      this->get_logger(), "[computePublish] Transforming output dataset from %s to %s.",
      cloud_tf->header.frame_id.c_str(), tf_output_frame_.c_str());
    // Convert the cloud into the different frame
    PointCloud2 cloud_transformed;
    if (!pcl_ros::transformPointCloud(tf_output_frame_, *cloud_tf, cloud_transformed, *tf_buffer_)) {
      RCLCPP_ERROR(
        this->get_logger(), "[computePublish] Error converting output dataset from %s to %s.",
        cloud_tf->header.frame_id.c_str(), tf_output_frame_.c_str());
      return;
    }
    cloud_tf.reset(new PointCloud2(cloud_transformed));
  }
  if (tf_output_frame_.empty() && cloud_tf->header.frame_id != tf_input_orig_frame_) {
    // no tf_output_frame given, transform the dataset to its original frame
    RCLCPP_DEBUG(
      this->get_logger(), "[computePublish] Transforming output dataset from %s back to %s.",
      cloud_tf->header.frame_id.c_str(), tf_input_orig_frame_.c_str());
    // Convert the cloud into the different frame
    PointCloud2 cloud_transformed;
    if (!pcl_ros::transformPointCloud(
          tf_input_orig_frame_, *cloud_tf, cloud_transformed, *tf_buffer_)) {
      RCLCPP_ERROR(
        this->get_logger(), "[computePublish] Error converting output dataset from %s back to %s.",
        cloud_tf->header.frame_id.c_str(), tf_input_orig_frame_.c_str());
      return;
    }
    cloud_tf.reset(new PointCloud2(cloud_transformed));
  }

  // Copy timestamp to keep it
  cloud_tf->header.stamp = input->header.stamp;

  // Publish a boost shared ptr
  pub_output_->publish(std::move(cloud_tf));
}
```

### Definition of done

Cloud data no more duplicated in `computePublish`

yukkysaito, 2022-05-31:
@VRichardJP thank you for your PR
Can I close this issue?

VRichardJP, 2022-05-31:
Yes

I think it would make sense to create a new task to track and fix similar problems across the rest of the code base. The filter one was maybe the most important since it is used a lot with big clouds, but I saw that a few other files have the same pattern:

```C++
auto tmp = PointCloud2();
someProcessing(in, tmp);
auto output = std::make_xxx<PointCloud2>(tmp); // copy 
```

Where the copy can always be avoided by using the correct pointer type for tmp intermediate cloud.

closed 2022-06-15 by BonoloAWF

esteve, 2022-06-22:
@VRichardJP

> but I saw that a few other files have the same pattern:

Do you remember what modules had this pattern? I want to make sure I'm covering all of them.

Thanks.

VRichardJP, 2022-06-23:
@esteve 
No idea if it is complete, but a grep on pattern `make_xxx<*PointCloud2>(xxx)` give me this list of files:

```
$ grep -rlIE "make_.*<.*PointCloud2>\([^)]+\)" src
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/concatenate_data/concatenate_data_nodelet.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/filter.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/vector_map_filter/lanelet2_map_filter_nodelet.cpp
src/universe/autoware.universe/sensing/probabilistic_occupancy_grid_map/include/laserscan_based_occupancy_grid_map/laserscan_based_occupancy_grid_map_node.hpp
src/universe/vendor/pointcloud_to_laserscan/src/pointcloud_to_laserscan_node.cpp
```