autowarefoundation/autoware_universe #1076: Error in obstacle avoidance direction when using extended lane to avoid obstacles

shulanbushangshu, 2022-06-10:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The default .yaml configuration for the extended lane in the obstacle avoidance module is shown below:
[image]
Wrong direction for obstacle avoidance when placing an obstacle in the left lane.As shown below：
[image]

When I change the default configuration：detection_area_left_expand_dist:1-->0,The above phenomena will not occur.

I think there is a problem with the code when using the extended lane for obstacle avoidance.

This phenomenon also occurs when we use relatively wide lanes (non-standard lanes) in reality

### Expected behavior

Whether extended lanes or wide lanes are used for obstacle avoidance, the direction of obstacle avoidance shall not be different.

### Actual behavior

Wrong direction for obstacle avoidance when placing an obstacle in the left lane by using default .yaml configuration for the extended lane in the obstacle avoidance module
[image]

### Steps to reproduce

1.Start autoware planning simulator using default;
2. Set initial position and gobal position;
3.Place an obstacle vehicle in the left lane

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

### Possible causes

First, confirm whether the code supports wide lane obstacle avoidance and extended Lane obstacle avoidance.
If it is not supported for design purposes, you need to change the default configuration file:detection_area_left_expand_dist:1-->0(Or delete relevant codes).

If the design needs support, I think there is a problem in the calculation of the obstacle avoidance module.

zulfaqar-azmi-t4, 2022-06-13:
@shulanbushangshu 
Yes, this indeed seems like a bug on the shift point computation.
I'll investigate them and let you know once I get the results.

shulanbushangshu, 2022-06-16:
As  show below：
[image]
The current obstacle avoidance strategy only considers whether the obstacle intersects with the lane boundary line; Instead of considering whether the boundary lines of obstacles and path intersect.
In my opinion, if the extended lane or wide lane is to be considered, whether the current strategy needs to be changed.

In order to adapt to the extended lane, the calculation method of obstacle avoidance offset distance may need to be changed too.
The current offset distance method is as follows:
[image]

@zulfaqar-azmi-t4 @taikitanaka3 
have any ideas

zulfaqar-azmi-t4, 2022-06-16:
As the mitigation, I was thinking of querying the whether the object overlaps with the object and if it not overlapping, the the object was ignore.

Currently the collision check might be a bit off. So I am investigating the proper solution to it.

zulfaqar-azmi-t4, 2022-06-16:
@shulanbushangshu
Hi, currently I'm using the following logic as the mitigation, but I am interested in knowing in case you have any concerns. Basically if the shift is shorter that the distance to the object, then we can safely ignore them.

[image]

closed 2022-06-27 by shulanbushangshu