autowarefoundation/autoware_universe #1087: Describe the origin of pre-trained models

miursh, 2022-06-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As mentioned [here](https://github.com/orgs/autowarefoundation/discussions/403), TIER IV member should describe the details about provided ML models.

### Purpose

Clarify what kind of dataset was used for training. 

### Possible approaches

Write the following information.

* If trained on a private dataset, at least some information on the nature of the data (e.g. camera images from Japanese traffic lights, lidar data from urban traffic in Europe...)
* training/testing set size, number of objects by types, etc
* test accuracy

### Definition of done

Complete the description of the following 5 nodes
- [ ] https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_apollo_instance_segmentation
- [ ] https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_centerpoint
- [ ] https://github.com/autowarefoundation/autoware.universe/tree/main/perception/traffic_light_ssd_fine_detector
- [ ] https://github.com/autowarefoundation/autoware.universe/tree/main/perception/tensorrt_yolo
- [ ] https://github.com/autowarefoundation/autoware.universe/tree/main/perception/traffic_light_classifier

miursh, 2022-07-13:
Update for tensorrt_yolo https://github.com/autowarefoundation/autoware.universe/pull/1339

ktro2828, 2022-07-20:
For lidar_appllo_instance_segmentation, there is already description of model in README.md

yukke42, 2023-02-17:
The document of centerpoint has updated in <https://github.com/autowarefoundation/autoware.universe/pull/2553>.

yukke42, 2023-02-17:
All of the task has completed.

closed 2023-02-17 by yukke42