autowarefoundation/autoware_universe #1102: The distortion_corrector has no warning when a velocity_report topic isn't subscribed

yukke42, 2022-06-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The distortion_corrector runs successfully even if a velocity report topic cannot be subscribed.

### Expected behavior

If the distortion_corrector node cannot subscribe a velocity_report topic, there should be a warning or fail to run.

### Actual behavior

same as the description

### Steps to reproduce

1. filter the `/vehicle/status/velocity_status` topic from rosbag
2. run `sensing` module

xmfcx, 2022-07-26:
@yukke42 any updates on this?

badai-nguyen, 2022-07-27:
@yukke42 Since the input status of new distortion_correction has already checked and logged [here.](https://github.com/autowarefoundation/autoware.universe/blob/156d0056b459d2dda0561718d4195c4ecd8960c8/sensing/pointcloud_preprocessor/src/distortion_corrector/distortion_corrector.cpp#L172)  
IMO, we don't need to add additional diagnostics for that. 
shall we close this issue? cc @miursh

yukke42, 2022-07-27:
@badai-nguyen Cloud you put the link to PR here? We can close this issue.

miursh, 2022-07-29:
Resolved by https://github.com/autowarefoundation/autoware.universe/commit/6fbdce5513b8a3992531da2c0a70597b4e293f2a

closed 2022-07-29 by miursh