autowarefoundation/autoware_universe #1105: Check Simulator's Compatibility with Autoware

Soohyeok-Park, 2022-06-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

- according to the issue discussed on simulation WG(https://github.com/autowarefoundation/autoware-projects/issues/21#issue-1203352244), MORAI Inc. has made a sample unit test package to check a simulator's compatibility with autoware.
- the package covers test cases from ControlModeCommand& ControlModeReport(no. 1) to HazardLightsCommand & HazardLightsReport(no. 6). 
- EmergencyState is not covered since it is not being determined yet.

### Purpose

- To ensure that any simulator is compatible with the latest version of Autoware
- Ultimately, streamlining the development process of Autoware-compatible simulators

### Possible approaches

A very simple method to check the compatibility of MORAI SIM: Drive with Autoware is included in this documentation, which can give a very brief idea at least.
[Link](https://morai.atlassian.net/wiki/spaces/AWF/pages/1107034224/Shared+MORAI+SIM+Drive+Quick+Start+Guide+for+AWF#4.-Running-Samples-Codes-to-Demonstrate-autoware_auto_msgs-Compatibility)

### Definition of done

[Current] cover the interface between the simulator and Autoware
[Future] test would cover other requirements defined in the documentation (related discussion [link](https://github.com/orgs/autowarefoundation/discussions/234))

sglee-morai, 2022-06-22:
1) Common Test Code = Manual Testing Code
2) Test Code for MORAI SIM: Drive
- Based on Common Test Code
- Test Automation Feature
  - pytest
  - MSC (MORAI SIM Control API): test setup / tear teardown method

sglee-morai, 2022-07-06:
# Update 22-07-06
Want to ask for a review: TierIV to test run the code.
However, I have a question regarding this, though. When a certain person (MORAI in this case) asks other members to run the code in their machine, how he or she can do that?

For me to have Soohyuk's code in my machine, I had to do the followings

open terminal
change working directory to ~/autoware/src/universe
change remote to Soohyuk's Autoware.Universe repository (which is a fork from the original Autoware.Universe repository)
git pull
git checkout [Soohyuk's latest commit]
Is this a typical process for other members as well in case wanting to get the code and run it in his/her local machine?

WJaworskiRobotec, 2022-07-06:
To remove morai msgs from the repo. 

I will test the code with TierIV Sim.

WJaworskiRobotec, 2022-07-15:
Tests work with TierIV Sim. I've approved the PR but there are some issues with CI tests.