autowarefoundation/autoware_universe #1109: Avoid point cloud copies

esteve, 2022-06-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

There are places in the code where a temporary point cloud is created and used to generate an output, which involves unnecessary copying. For context, see https://github.com/autowarefoundation/autoware.universe/issues/998

### Purpose

Avoid unnecessary copying of point clouds.

### Possible approaches

Depends on each code block where this issue is happening.

### Definition of done

Processing of point clouds is done in place instead of copying.

mitsudome-r, 2022-06-21:
We should also use pointcloud_wrapper like in Autoware.Auto to have more efficient pointcloud manipulation.

mitsudome-r, 2022-07-05:
We would like to investigate if there are unnecessary copies in the following files ([link](https://github.com/autowarefoundation/autoware.universe/issues/998#issuecomment-1163829806))

```
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/concatenate_data/concatenate_data_nodelet.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/filter.cpp
src/universe/autoware.universe/sensing/pointcloud_preprocessor/src/vector_map_filter/lanelet2_map_filter_nodelet.cpp
src/universe/autoware.universe/sensing/probabilistic_occupancy_grid_map/include/laserscan_based_occupancy_grid_map/laserscan_based_occupancy_grid_map_node.hpp
src/universe/vendor/pointcloud_to_laserscan/src/pointcloud_to_laserscan_node.cpp
```

esteve, 2024-03-27:
Closing as this is a very old issue.

closed 2024-03-27 by esteve