autowarefoundation/autoware_universe #1115: Motion Utils Package in the Universe

purewater0901, 2022-06-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Creating a new package called motion utils under the common directory. It contains common functions used in planning and control packages. 

### Purpose

Since there are many common functions in planning and control packages in the universe, we need one common package that contains those duplicated functions. Currently, tier4_autoware_utils has some useful functions to use many packages, but it does not have any functions only used in planning and control modules.

### Possible approaches

Creating a common utils package under the common directory.

### Definition of done

Creating a package that contains common functions for planning and control packages.

purewater0901, 2022-06-16:
@TakaHoribe @yukkysaito @kenji-miyake 
What do you think about this idea? I also wonder if you have better names for the motion utils package.

yukkysaito, 2022-06-16:
How about creating trajectory utils, etc. in tier4_autoware_utils?
Can you tell us more details about the common process for planning and control?

purewater0901, 2022-06-16:
Currently, there are too many functions in tier4_autoware_utils, we'd like to separate some functions exclusive to planning and control modules. For example, I'd like to add a function called insertStopPoint which inserts a stop point to a trajectory, but this function is only used for planning and control modules. 

Therefore, I think creating a whole new package would be better .

yukkysaito, 2022-06-16:
How do you think about the separation between [here](https://github.com/autowarefoundation/autoware.universe/blob/main/common/tier4_autoware_utils/include/tier4_autoware_utils/trajectory/trajectory.hpp) and new package ?
This package uses only for planning and control.

purewater0901, 2022-06-16:
@yukkysaito Sorry, I need to give you more details. I'm going to add [these functions](https://github.com/autowarefoundation/autoware.universe/blob/main/common/tier4_autoware_utils/include/tier4_autoware_utils/trajectory/trajectory.hpp) to a new package as well.

yukkysaito, 2022-06-16:
@purewater0901 Thanks. 
I agree with your suggestion

TakaHoribe, 2022-07-11:
@xmfcx @mitsudome-r @angry-crab We are cleaning up the motion-related common utility functions. There are now many packages with duplicated functions:
[autoware_auto_common](https://github.com/autowarefoundation/autoware.universe/tree/main/common/autoware_auto_common)
[autoware_auto_geometry](https://github.com/autowarefoundation/autoware.universe/tree/main/common/autoware_auto_geometry)
[motion_common](https://github.com/autowarefoundation/autoware.universe/tree/main/common/motion_common)
[tier4_autoware_utils](https://github.com/autowarefoundation/autoware.universe/tree/main/common/tier4_autoware_utils)

Any comments, concerns, or suggestions?

angry-crab, 2022-07-15:
Ideally, we would like to merge the packages into a single one. However, some packages are from `Autoware.Auto`, and are not likely to be maintained in the future, but they might be required if we decide to port more packages from `Auto`. In this case, it might not be a good idea to merge them. I think breaking down `tier4_autoware_utils` as discussed above is okay, or we could have something like `autoware_motion_extension` for further development.

kenji-miyake, 2022-09-19:
I guess we can close this.
https://github.com/autowarefoundation/autoware.universe/tree/5f7fbd74171ef90b5f84c35e9e0088bb8f71690e/common/motion_utils

@purewater0901 Please update the status.

takayuki5168, 2022-10-25:
I close the issue since motion_utils package has been created and used in autoware.universe.

closed 2022-10-25 by takayuki5168

purewater0901, 2022-12-22:
@TakaHoribe What do you think about this change?