autowarefoundation/autoware_universe #1116: Obstacle stop planner is chattering in obstacle detection by pointcloud and its marker continue visualizing

h-ohta, 2022-06-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

There are two problems.
- When blinking pointcloud, obstacle detection is chattering.
- When chattering is occured, but wall marker keeps visualizing.

https://user-images.githubusercontent.com/9547070/173994863-3cae344c-584e-4343-9c75-75060ceb4b26.mp4

### Expected behavior

When blinking pointcloud, obstacle detection is not chattering. 
Detecction result keeps for the user defined time.
I believe that the wall marker synchronize with its detection result.

### Actual behavior

When blinking pointcloud, obstacle detection is chattering.
When chattering is occured, but marker keeps visualizing

### Steps to reproduce

1. Drive ego vehicle
2. set the obstacle in front of ego
3. move away it immediately
4. set the obstacle again immediately

like this video

https://user-images.githubusercontent.com/9547070/173995739-9c4487c6-8a7e-44e6-bdbd-58dc6de83948.mp4

h-ohta, 2022-06-20:
@takayuki5168 @satoshi-ota Could you confirm this?

takayuki5168, 2022-06-21:
Thanks for your report. As you reported, I've encountered this behavior sometimes.

h-ohta, 2022-08-31:
This issue is resolved by #1513 and #1458

closed 2022-08-31 by h-ohta