autowarefoundation/autoware_universe #1121: The `detection_area` module does not visualize debug markers when some obstacles are not detected by it

h-ohta, 2022-06-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Now, the `detection_area` module visualizes  no debug marker when some obstacles are not detected.

### Expected behavior

 I can decide whether obstacles are not detected, module is unregistered or ego vehicle goes over the dead line from debug markers.

### Actual behavior

I can't decide detection area behavior from debug markers.

### Steps to reproduce

1. drive ego vehicle
2. put a obstacle in detection area
3. move away it from detection area

https://user-images.githubusercontent.com/9547070/174035617-bdd27485-13ea-4fde-8de7-55ce1d6821dc.mp4

h-ohta, 2022-06-20:
@kyoichi-sugahara @TakaHoribe Could you confirm this?

kyoichi-sugahara, 2022-06-22:
@h-ohta 
I think your proposal is reasonable.
(and also explanation about dead_line in READMED needs to be added...)

sujithvemi, 2022-07-12:
We have tried to reproduce the issue. But detection area is not at all coming for us after enabling DetectionArea in Rviz. Are the steps to reproduce above sufficient? Or are there any other steps missing in the details? Is it expected to be shown while running the planning tutorial given in the documentation or should we run some other launch files for this?

raminterpl, 2022-07-18:
@h-ohta : can you pls reply to sujith

h-ohta, 2022-07-19:
@raminterpl @sujithvemi Sorry for late reply.
Could you tell me your testing environment ?
I can confirm it.

sujithvemi, 2022-07-19:
I am using RViz and starting the simulation through the planning simulator launch file provided in the tutorial. I am running it through docker environment.

h-ohta, 2022-07-20:
@sujithvemi I see. Could you tell me your operating procedure at detail ?

h-ohta, 2022-07-20:
Or I'll recheck my issue report and rewrite it. Just a moment...

h-ohta, 2022-07-20:
I found that the map, used in the planning simulator tutorial, does not have detection area...
I'll think other resolutions.

h-ohta, 2022-07-20:
@sujithvemi I I made new map which has detection area and updated issue report. Could you try it?

sujithvemi, 2022-07-31:
Hi @h-ohta ,

Sorry for the late reply, my laptop had some motherboard issue and was sent for service, I just got it couple of days ago. 

I have tried it with the new map and the behavior is as shown in the video, detection area markers are only shown when the obstalce is in the specified area. You can verify that in the video attached. Also, my colleague has observed the same in an independent test. Kindly ignore the slow performance, running some other things on the laptop, so the detection area colour reversal took some extra time.

https://user-images.githubusercontent.com/35031465/182028294-cbd1c38f-6786-45e4-9073-436e35f8d177.mp4

xmfcx, 2022-09-06:
So we are expecting the area to be published even when there is no obstacle, but in different colors.

> My proposal is
> module is registered but not detected: detection area marker is green
> detected: detection area marker is red
> go over the dead line: detection area marker is grey
> unregistered: delete

raminterpl, 2022-09-13:
@xmfcx :Status: My team is working on this.

mitsudome-r, 2022-09-27:
@raminterpl Do you have any updates on this issue?

sujithvemi, 2022-10-09:
Sorry for the late reply. I had identified the code that needs to be changed, just working on the way to implement it. That is because the way the code is currently written is not modular to implement the requirements mentioned here, color and other attributes are provided in the function directly and are not populated using a variable or conditionally, that's why need to change more than just add a block of markers. Will update in a day or two about the final implementation details.

sujithvemi, 2022-10-31:
I am working on the changes, I was occupied with some other work. Will commit a code for review by tomorrow.

sujithvemi, 2022-10-31:
Hi, I am not able to push the code. I am getting some authentication error, is there any credential requirement that need to be satisfied to push the code? I am using my personal credentials (the one with which I am interacting here).

raminterpl, 2022-10-31:
@BonoloAWF @mitsudome-r : Please add sujith email id: [sujith.3920@gmail.com](mailto:sujith.3920@gmail.com)  In the allowed autoware git users commit list

BonoloAWF, 2022-10-31:
@sujithvemi invite for the team is sent, please accept.

sujithvemi, 2022-11-06:
@BonoloAWF @mitsudome-r I am trying to push a new branch to the sub repo Autoware.Universe. But I am getting the error shown in the below image. I was first getting the password authentication removed error first, so I setup a Personal Access Token (PAT) from the guidelines given for Github and tried to push, then I was receiving this error. Can you please help me out with what I am missing here.
[image]

xmfcx, 2022-11-06:
@sujithvemi Please fork the autoware.universe and push your branch there. Then you can create a PR from the branch of your fork.

For more info: https://autowarefoundation.github.io/autoware-documentation/main/contributing/pull-request-guidelines/#general-pull-request-workflow

closed 2023-01-11 by xmfcx