autowarefoundation/autoware_universe #1127: The LiDAR's azimuth unit is less understandable

badai-nguyen, 2022-06-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, the LiDAR's azimuth is output in hundredths of degree unit that makes the code being less understandable and error-prone when applying for new LiDAR.
The bug was occurred when applied for a new LiDAR here issue #509 

### Purpose

Improve the code readability and application for new LiDAR vendor

### Possible approaches

- Change output azimuth unit to degree in range of [0.0f , 360.0f] or radian in range of [ 0.0f , 2*pi] in LiDAR drivers: [velody ](https://github.com/tier4/velodyne_vls) and [hesai](https://github.com/tier4/hesai_pandar)
- change the respectively unit of azimuth referenced in the related repository.  
  - [dual_return_outlier_filter ](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/src/outlier_filter/dual_return_outlier_filter_nodelet.cpp)
  - [ring_outlier_filter](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp)
  - [LiDAR blockage_diagnostics](https://github.com/autowarefoundation/autoware.universe/tree/main/sensing/pointcloud_preprocessor/src/blockage_diag)

The degree unit seems easier for readers to image the value of angle setting while radian unit maybe save the conversion for sinusoidal/co-sinusoidal related function. Since the main purpose is for readability,  I suppose to use degree for azimuth.  

### Definition of done

Unify the new unit of azimuth in AW. 
- [ ] output azimuth unit change from velody driver
- [ ] output azimuth unit change from hesai driver
- [ ] input azimuth unit change for related code and parameter in universe

xmfcx, 2022-11-15:
My suggestion: Following the [discussions for the new point type `PointXYZIRCT`](https://github.com/autowarefoundation/autoware-documentation/pull/180#issuecomment-1315272014) I think we can do the following:

- Don't calculate the azimuth in the driver.
- Calculate it with `p.azimuth = std::atan2(p.y, p.x)` in the place it is necessary
- The output will be a `double` or `float` (whatever precision is needed) in the range [-π , +π] radians 
- Works for any kind of lidars, including solid state lidars too.

xmfcx, 2022-11-15:
@xmfcx check for how long it takes to compute std::hypot and std::atan2 for large amount of points just to see how much computation it takes.

Also list what nodes make use of azimuth and distance fields.

@esteve also suggests to send down the values because storage/memory is cheaper.

xmfcx, 2022-11-16:
I've created a mini benchmark for this topic: https://github.com/xmfcx/bench_point_cloud#readme @esteve @miursh 

Let's continue this discussion in https://github.com/autowarefoundation/autoware-documentation/pull/180#issuecomment-1317583569

closed 2022-11-16 by xmfcx