autowarefoundation/autoware_universe #1147: radar_tracks_msgs_converter Build Error

MichaelShen97, 2022-06-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fail to build in docker because fail to find header file `radar_msgs/msg/radar_tracks.hpp`.

### Expected behavior

Successful build.

### Actual behavior

`--- stderr: radar_tracks_msgs_converter
In file included from /home/mh/software/autoware/src/universe/autoware.universe/perception/radar_tracks_msgs_converter/src/radar_tracks_msgs_converter_node/radar_tracks_msgs_converter_node.cpp:15:
/home/mh/software/autoware/src/universe/autoware.universe/perception/radar_tracks_msgs_converter/include/radar_tracks_msgs_converter/radar_tracks_msgs_converter_node.hpp:27:10: fatal error: radar_msgs/msg/radar_tracks.hpp: No such file or directory
   27 | #include "radar_msgs/msg/radar_tracks.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/radar_tracks_msgs_converter_node_component.dir/build.make:63: CMakeFiles/radar_tracks_msgs_converter_node_component.dir/src/radar_tracks_msgs_converter_node/radar_tracks_msgs_converter_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/radar_tracks_msgs_converter_node_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< radar_tracks_msgs_converter [2.09s, exited with code 2]
Aborted  <<< sample_sensor_kit_launch [0.71s]
Aborted  <<< tier4_vehicle_rviz_plugin [1.80s]
Aborted  <<< tier4_autoware_api_launch [1.89s]                                                                                                                                                
Aborted  <<< lanelet2_map_preprocessor [2.20s]                                                                                                                                               
Aborted  <<< grid_map [2.07s]                                                                                                                                                                
Aborted  <<< occupancy_grid_map_outlier_filter [42.1s]                                                                                                                                          
Aborted  <<< tier4_traffic_light_rviz_plugin [48.7s]                                                                                                                                      
Aborted  <<< ground_segmentation [57.0s]                                                                                                                                                  
Aborted  <<< compare_map_segmentation [57.2s]                                                                                                                                                    
Aborted  <<< accel_brake_map_calibrator [56.3s]                                                                                                                                             
Aborted  <<< obstacle_cruise_planner [57.5s]                                                                                                 
Aborted  <<< behavior_path_planner [1min 31s]                                             

Summary: 190 packages finished [1min 47s]
  1 package failed: radar_tracks_msgs_converter
  12 packages aborted: accel_brake_map_calibrator behavior_path_planner compare_map_segmentation grid_map ground_segmentation lanelet2_map_preprocessor obstacle_cruise_planner occupancy_grid_map_outlier_filter sample_sensor_kit_launch tier4_autoware_api_launch tier4_traffic_light_rviz_plugin tier4_vehicle_rviz_plugin
  5 packages had stderr output: compare_map_segmentation ground_segmentation occupancy_grid_map_outlier_filter radar_tracks_msgs_converter tier4_traffic_light_rviz_plugin
  5 packages not processed
`

### Steps to reproduce

Simply follow the [official instructions on docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/).
1. git clone the latest version `git clone https://github.com/autowarefoundation/autoware.git && cd autoware`
2. Install NVIDIA docker following [official instructions](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#docker).
3. Install rocker with `sudo apt install python3-rocker`
4. `rocker --nvidia --x11 --user --volume {path/to/autoware} -- ghcr.io/autowarefoundation/autoware-universe:latest`, where `{path/to/autoware}` is substituted with the actual path.
5. `mkdir src && vcs import src < autoware.repos`
6. `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`
The error occurs and the build terminated.

### Versions

- OS: Ubuntu 20.04.4
- ROS2: galactic
- Autoware: latest

kosuke55, 2022-06-22:
@MichaelShen97 
please run `rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO` in autoware dir.

BonoloAWF, 2022-06-22:
Hi @MichaelShen97 the best way to get help with Autoware is posting in the Q&A section of the Discussions here: https://github.com/autowarefoundation/autoware/discussions/categories/q-a

closed 2022-06-22 by BonoloAWF