autowarefoundation/autoware_universe #1151: Non-successive way points generated by A-star algorithm in freespace planner

NorahXiong, 2022-06-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Way points generated by A-star algorithm are not successive as the red circle shows. It could result in bad trajectory control.

[image]

### Expected behavior

Way points generated by A-star algorithm are successive. Here successive means the vehicle can move from the former point to the next point through a movement in the transition_table_. Possible transitions are defined as below.
[AstarSearch::TransitionTable createTransitionTable function](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/freespace_planning_algorithms/src/astar_search.cpp#L74)

### Actual behavior

Way points generated by A-star algorithm are not successive.

### Steps to reproduce

1. Start autoware planning simulator using default
2. Set pose estimate and goal pose the parking lot

(These steps can not guarantee the problem will repeat. According to my experience, there is a higher probability if the number of arcs in ReesShepp solution is large.)

### Versions

- OS: Ubuntu 20.04
- ROS2: Galactic
- Autoware: Autoware.universe:main

### Possible causes

One node in close list is rewrite during the search, however the connected node stays the same. If the preceding node changed, the nodes behind should recheck the transition relation and recalculate the actual cost. But these steps are not implemented now.

For example. In the following picture, S is the start point, G is the goal point and P(P*) is the non-successive point. At first, nodes P* and its child D to G are all in close list. Point C is chosen from the open list and is under check. Node P is the candidate transition target of C and after check point P is selected as the child of C. Actually P and P* are the same node in search space cause they are in the same grid and have the same orientation. In other words, P and P* use the same part of memory and share the same index in search space. Even though P is not totally equal to P*, D keeps pointing to P/P*. If a better route from P to G is found, the problem may not appear. If not, S-C-P-D-G is selected as the best way, and we get a non-successive way point array.

[image]

mitsudome-r, 2022-06-28:
@TakaHoribe Could you assign someone from Planning team to take a look at this?

NorahXiong, 2022-08-03:
The bug is caused by queuing a node more than once and modifying the node elements for different queues. Such operation results in incorrect data association and exploded when retrieving a queue to generate the final path.  It can be fixed by creating an new node instance for data storage every time the node is queued.  Another strategy is just to forbid a node to be queued second time just as the implementation in [ROS-Nav2](https://github.com/ros-planning/navigation2/blob/galactic/nav2_smac_planner/src/a_star.cpp#L372). 

I tried to fix it by the first method but the node data storage implementation [needs alteration](https://github.com/orgs/autowarefoundation/discussions/2716) as we don't know exactly how many times will a node be queued. I simply substituted the pre-allocation by instant-allocation and the memory consumption obviously decreased and meanwhile the time consumption obviously increased. In general, the performance is not very good. Maybe some other container such as map could improve the instant-allocation performance.

So instead I used the second method with reference to ROS-Nav2 implementation even though it will block some possible paths. I tested several cases after the fix and it works for them all. Maybe the tiny constraint of search space has little influence on the result because the visit to search space is originally very sparse in most cases. In ROS-Nav2 implementation, some tricks as [AnalyticExpansion](https://github.com/ros-planning/navigation2/blob/galactic/nav2_smac_planner/src/a_star.cpp#L471) are composed into the main algorithm and I think it could benefit the algorithm to some extent. 

If there are better solutions, please let me know.

PR #1504 (2022-09-06): fix(freespace_planning_algorithms): non-successive waypoints problem in hybrid A* algorithm

## Description

fix #1151 

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes b [... cut]
closed 2022-09-06 by xmfcx