autowarefoundation/autoware_universe #1160: `<arg>` is overwritten when including multiple launch files because `<include>` does not scope

h-ohta, 2022-06-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

We probably have same issue as https://github.com/ros2/launch/issues/620.

### Purpose

We have to fix above issue to avoid unexpected error in the future.

### Possible approaches

add `<group>` tag around `<include>` tag is better.

### Definition of done

Now, I only confirm and fix in tier4_xxx_launch directory.
- [ ] tier4_autoware_api_launch
- [ ] tier4_control_launch
- [ ] tier4_integration_launch
- [ ] tier4_localization_launch
- [ ] tier4_perception_launch
- [ ] tier4_planning_launch
- [ ] tier4_sensing_launch
- [ ] tier4_simulator_launch
- [ ] tier4_system_launch
- [ ] tier4_vehicle_launch

angry-crab, 2022-06-29:
If there is only one `include` in a launch file, is `group` tag still necessary?

angry-crab, 2022-07-05:
Recreated PRs
https://github.com/autowarefoundation/autoware.universe/pull/1235
https://github.com/autowarefoundation/autoware.universe/pull/1236
https://github.com/autowarefoundation/autoware.universe/pull/1237
https://github.com/autowarefoundation/autoware.universe/pull/1238
https://github.com/autowarefoundation/autoware.universe/pull/1239
https://github.com/autowarefoundation/autoware.universe/pull/1240

h-ohta, 2022-07-08:
> If there is only one include in a launch file, is group tag still necessary?

Sorry for late reply. It would be better.

angry-crab, 2022-07-13:
All related PRs are merged. I think we can close this issue. The unchecked packages are good.

h-ohta, 2022-07-13:
I've confirmed. LGTM

h-ohta, 2022-07-13:
@angry-crab Thank you for your cooperation!

closed 2022-07-13 by h-ohta