autowarefoundation/autoware_universe #1168: Planning Simulation cannot send engage signal to

brian841102, 2022-06-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In the planner simulation, only using terminal command to send engage signal can start the vehicle.
The AutowareStatePanel Button is malfunctioning.

### Expected behavior

When **Engage** Button was pressed, the vehicle started to move.

### Actual behavior

When **Engage** Button was pressed, the vehicle do not move.
AutowareStatePanel shows WAITING_FOR_ROUTE.
ad_service_state_monitor shows “MAP msg is not ready yet”
![Uploa
[image]
ding 截圖 2022-06-25 16.36.25.png…]()

``__

### Steps to reproduce

1. `ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/Downloads/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
`
2. add AutowareStateControl Panel
3. set initial pose & set goal pose
4. Press engage button on the panel

### Possible causes

ad_service_state_monitor cannot subscribe to the vector_map.

shmpwk, 2022-06-25:
This issue came from my commit. 
https://github.com/autowarefoundation/autoware.universe/pull/1153/files#diff-f092ba8cc68e325209a916a7446c37e3efb13e03313408d4461d05287542367eR462-R464

Thank you for notifying me.

brian841102, 2022-06-25:
> This issue came from my commit. https://github.com/autowarefoundation/autoware.universe/pull/1153/files#diff-f092ba8cc68e325209a916a7446c37e3efb13e03313408d4461d05287542367eR462-R464
> 
> I apologize my lacking of check and thank you for your contribution.

Not a big deal, don't mind.
I am new to the community and autoware.universe.
Although this is a small fix, this is also my first contribution.
I am pleased to do so and look forward to engaging more in the future.

closed 2022-06-25 by brian841102

kenji-miyake, 2022-06-25:
@brian841102 Thank you for your first contribution! :smile: 
And thank you @shmpwk @wep21 @maxime-clem for your quick support. :heart: 
I feel the Autoware community is getting better. :blush: