autowarefoundation/autoware_universe #1207: Wrong Drivable Area might be created for behavior path planner

taikitanaka3, 2022-07-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

 - cv::fillpoly sometimes does not cover area where there is an overwarping polygon
https://github.com/autowarefoundation/autoware.universe/blob/791fd51ea0d43df2796fac8d00df770c67ffd5be/planning/behavior_path_planner/src/utilities.cpp#L1161

see ref)
https://github.com/taikitanaka3/cpp_snippets/issues/1

### Expected behavior

fill poly can fill the area which has overwraping area 

### Actual behavior

However sometimes cv::fillpoly can't cover overwrapping area

### Steps to reproduce

tbd.

shmpwk, 2022-07-01:
It seems Open CV cannot avoid even-odd rule, and thus some people seems SVG [polyline](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/polyline).
https://github.com/opencv/opencv/issues/5256#issuecomment-954321603

taikitanaka3, 2022-07-01:
I am going to close this issue because this case is already handled in avoidance module

closed 2022-07-01 by taikitanaka3