autowarefoundation/autoware_universe #1228: Image messages are being dropped or coming with latency

kaancolak, 2022-07-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

 We are using the CycloneDDS configuration provided by @takam5f2 in this [discussion](https://github.com/autowarefoundation/autoware/discussions/2636) with tuned DDS advised by the ROS2 page. Without running Autoware.Universe, we are getting 20 fps with 3 cameras without any problem. But if we run the entire Autoware.Universe pipeline, image messages drop to ~3,4 FPS. When we reduce the resolution of the images to around 600x500 and fix the fps to 15, there is no frame loss, but the latency in the topic is around ~70,80 ms. Is there anyone have any advice or solution?
 
 CycloneDDS supports shared memory with ROS2 Humble. is there any plan to convert autoware messages to a fixed size for using zero-copy DDS operation with shared memory.

### Expected behavior

Image message topics must arrive with minimal latency.

### Actual behavior

Image messages are being dropped or coming with latency

### Steps to reproduce

1. Run Autoware with a 2D object detection pipeline.

mitsudome-r, 2022-07-05:
Possible relevant discussions: 
* https://github.com/orgs/autowarefoundation/discussions/2617
* https://github.com/orgs/autowarefoundation/discussions/2621
* https://github.com/orgs/autowarefoundation/discussions/2555

Also, as @xmfcx mentioned, if we put all image processing nodes in a single component, we should be able to reduce the image latency.

mitsudome-r, 2022-07-12:
We need a rosbag to reproduce this issue.

MertClk, 2022-07-18:
@Zeysthingz can you share bag files to reproduce this issue?

xmfcx, 2022-07-26:
@xmfcx share the bag files.

mitsudome-r, 2022-08-02:
@xmfcx Is the bag file ready?

xmfcx, 2022-08-16:
It seems all present messages were recorded instead of just the sensor messages. I will extract the sensor messages using this tool https://github.com/Kettenhoax/ros2bag_tools and upload the bag file.

mitsudome-r, 2022-09-13:
@xmfcx is investigating this issue.

xmfcx, 2022-09-19:
The shared bag files are/will be available in autoware documentation through this pr: https://github.com/autowarefoundation/autoware-documentation/pull/200

mitsudome-r, 2022-09-27:
Note: the bag file only contains velodyne_packets so needs conversion to pointcloud.

mitsudome-r, 2022-09-27:
@xmfcx could you share the link to sensor description repository for your vehicle?

xmfcx, 2022-10-04:
https://github.com/vautonomous/isuzu_sensor_description @xmfcx also add this to the documentation.

mitsudome-r, 2022-10-11:
From the discussion from LeoDrive's update in ASWG, the latency is likely to be caused by CPU usage. We might not be able to fix this within Bus ODD timeline.

mitsudome-r, 2022-10-11:
Using Iceoryx zero copy improved the message transportation, but using Iceoryx as AWF projects needs further discussion.

kaancolak, 2022-10-19:
In the relevant PRs, I passed the same `pointcloud_container` to Velodyne nodes and missed the occupancy grid map package. When working with `sample_sensor_kit` and  `logging_simulator.launch.xml/autoware.lauch.xml`, it decreased network traffic by around -%30 and also CPU utilization decreased.

 I will test it with a real Velodyne device, making sure Velodyne drivers work without any problem in the same container with a different namespace.

xmfcx, 2022-11-01:
Once
- https://github.com/autowarefoundation/autoware.universe/pull/2114
- https://github.com/autowarefoundation/sample_sensor_kit_launch/pull/48

are merged, we can close this issue.

BonoloAWF, 2022-11-07:
> Once
> 
>     * [feat(tier4_localization_launch): pass pc container to localization #2114](https://github.com/autowarefoundation/autoware.universe/pull/2114)
> 
>     * [feat(sample_sensor_kit_launch): pass container to velodyne nodes sample_sensor_kit_launch#48](https://github.com/autowarefoundation/sample_sensor_kit_launch/pull/48)
> 
> 
> are merged, we can close this issue.

PRs are ongoing, milestone updated.

closed 2022-11-14 by mitsudome-r