autowarefoundation/autoware_universe #1231: surround obstacle checker rarely dies in local environment with segmentation fault

satoshi-ota, 2022-07-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

`surround obstacle checker` rarely dies (1~2%) in local environment with segmentation fault.
Core dump file is currently being prepared now...

### Expected behavior

`surround obstacle checker` never dies with any situation.

### Actual behavior

`surround obstacle checker` rarely dies (1~2%) in local environment with segmentation fault.

### Steps to reproduce

1. enable `surround obstacle checker` [here](https://github.com/tier4/autoware_launch/blob/570e640b21b6b1916695d0b522df8be7ff2cb37c/planning_launch/launch/scenario_planning/lane_driving/motion_planning/motion_planning.launch.py#L262)
2. launch `planning_simulator.launch.xml`
3. set ego pose and goal
4. set object arround the ego

mitsudome-r, 2022-07-12:
@satoshi-ota Are there any suggested placement of the obstacles to reproduce the error? Also, will you be able to do the investigation yourself or would you like someone else to help you.

satoshi-ota, 2022-07-12:
> @satoshi-ota Are there any suggested placement of the obstacles to reproduce the error? Also, will you be able to do the investigation yourself or would you like someone else to help you.

@mitsudome-r I guess that there is no specific condition about the obstacle placement, but this error is caused in following scenario test (the obstacle is always same placement in scenario test).

- UC-001-0002_PSim_1_case2_v2

I tried but couldn't fix this error, so I want help... :pleading_face: 

cc @0x126 @TomohitoAndo

VRichardJP, 2022-07-16:
I have not looked at the code, but what about the target frame being an empty string?
```
#6  0x00007fc03692a2be in tf2_ros::Buffer::lookupTransform (this=0x7fc014004490, target_frame="", source_frame="base_link", time=..., timeout=...)
    at /opt/ros/galactic/include/tf2_ros/buffer.h:104
#7  0x00007fc036926e0d in surround_obstacle_checker::SurroundObstacleCheckerNode::getTransform (this=0x7fc014004110, source="", target="base_link", stamp=..., 
    duration_sec=0.5) at /home/autoware/git/pilot-auto.x2/src/autoware/universe/planning/surround_obstacle_checker/src/node.cpp:409
```

VRichardJP, 2022-07-16:
I have not checked the code but could it be the empty string as input frame being undefined behavior? What if you prevent the lookupTransform when the frame is empty?

VRichardJP, 2022-07-16:
Something is strange with the validateFrameId function, no?
https://github.com/ros2/geometry2/blob/1186ee1e1be1dfe3691f31ba7f13662b1f1e1b91/tf2/src/buffer_core.cpp#L162

There is an early return in the case the frame_id is empty. And yet in your case it manages to reach the lookupFrameNumber function. So is there a crash on the lookup of "base_link"? or could it be the function is aggressively optimized because the frame_id is assumed not to be empty?

satoshi-ota, 2022-07-19:
Thanks for your comments. 
Yes, as you said, the function is suspicious. There is an early return in the function, so I thought we don't have to guard empty frame id at outside the lookupTransform, but...

> So is there a crash on the lookup of "base_link"? or could it be the function is aggressively optimized because the frame_id is assumed not to be empty?

I can't deny the both possibility.

mehmetdogru, 2022-08-02:
@satoshi-ota 

I could not reproduce the error, so could you share the scenario `UC-001-0002_PSim_1_case2_v2` and the lanelet2 map used?

satoshi-ota, 2022-08-18:
@mehmetdogru cc @mitsudome-r @TakaHoribe 

Thanks for your contribute :+1: But unfortunatelly, I can't share the lanelet2_map because of lisence :bow: 
I guessed that this strange bug maybe cause no matter whether we use scenario sim or not. So, I trying to reproduce this error in my local environment without scenario simulator. (but I could not reproduce the error, too. :pleading_face:)

mitsudome-r, 2022-08-23:
Removing @mehmetdogru from assignee until there is a reproducing procedure available in public.

mitsudome-r, 2022-09-13:
@satoshi-ota  Is anyone from TIER IV working on this? Since the scenario file to reproduce the error is only accessible by TIER IV, we cannot assign anyone from AWF.

mitsudome-r, 2022-09-27:
I have removed high priority tag since we cannot assign anyone unless we have a way to reproduce the issue.

mitsudome-r, 2022-09-27:
@satoshi-ota 
@satoshi-ota friendly ping.

> @satoshi-ota Is anyone from TIER IV working on this? Since the scenario file to reproduce the error is only accessible by TIER IV, we cannot assign anyone from AWF.