autowarefoundation/autoware_universe #1247: `behavior_velocity_planner` sometimes crashes before intersection

VRichardJP, 2022-07-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Yesterday, during test drive on private site, I observed `behavior_velocity_planner` crashes often when approaching an intersection. The crash cause is always an out-of-bound access `input_path.points.at(stop_line_idx)` here:
https://github.com/autowarefoundation/autoware.universe/blob/62476712f16cc61c0a6fc08b10f518554f1fbab0/planning/behavior_velocity_planner/src/scene_module/intersection/scene_intersection.cpp#L163

### Expected behavior

No crash

### Actual behavior

crash

### Steps to reproduce

1. Run behavior_velocity_planner on a lanelet map with intersections (manual driving is ok, simulator maybe too)

### Versions

- Ubuntu 20.04
- galactic
- private fork of main

### Possible causes

To dig further, I recompiled `behavior_velocity_planner` module in Debug mode and ran it with gdb. Here is the full backtrace when the crash occurs:
```
[New Thread 0x7fff9907c700 (LWP 702884)]
[Thread 0x7fff9907c700 (LWP 702884) exited]
[INFO] [1657025745.719852065] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: ===== plan end =====
[INFO] [1657025745.720560310] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: ===== plan start =====
[INFO] [1657025745.720761729] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: lane_id = 764, state = GO
[INFO] [1657025745.723838402] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: stop line line is at path[0], ignore planning.
[INFO] [1657025745.723862781] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: ===== plan end =====
[INFO] [1657025745.724849081] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: ===== plan start =====
[INFO] [1657025745.725049964] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: lane_id = 765, state = GO
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 81) >= this->size() (which is 81)
--Type <RET> for more, q to quit, c to continue without paging--

Thread 10 "component_conta" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe67fc700 (LWP 700468)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff793f859 in __GI_abort () at abort.c:79
#2  0x00007ffff7bca911 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7bd638c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff7bd63f7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7bd66a9 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7bcd3ab in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fffca949fb4 in std::vector<autoware_auto_planning_msgs::msg::PathPointWithLaneId_<std::allocator<void> >, std::allocator<autoware_auto_planning_msgs::msg::PathPointWithLaneId_<std::allocator<void> > > >::_M_range_check(unsigned long) const (this=0x7fffe67f8838, __n=81)
    at /usr/include/c++/9/bits/stl_vector.h:1070
#8  0x00007fffca95540f in std::vector<autoware_auto_planning_msgs::msg::PathPointWithLaneId_<std::allocator<void> >, std::allocator<autoware_auto_planning_msgs::msg::PathPointWithLaneId_<std::allocator<void> > > >::at(unsigned long) const
    (this=0x7fffe67f8838, __n=81) at /usr/include/c++/9/bits/stl_vector.h:1109
#9  0x00007fffcaac6666 in behavior_velocity_planner::IntersectionModule::modifyPathVelocity(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*, tier4_planning_msgs::msg::StopReason_<std::allocator<void> >*)
    (this=0x7fffb40320a0, path=0x7fffe67f9540, stop_reason=0x7fffe67f9200)
    at /home/sig/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner/src/scene_module/intersection/scene_intersection.cpp:189
#10 0x00007fffca978653 in behavior_velocity_planner::SceneModuleManagerInterface--Type <RET> for more, q to quit, c to continue without paging--
::modifyPathVelocity(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*) (this=0x5555555bd070, path=0x7fffe67f9540)
    at /home/sig/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner/include/scene_module/scene_module_interface.hpp:162
#11 0x00007fffca9795c4 in behavior_velocity_planner::SceneModuleManagerInterfaceWithRTC::plan(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*) (this=0x5555555bd070, path=0x7fffe67f9540)
    at /home/sig/autoware/src/universe/autoware.universe/planning/behavior_velocity_planner/include/scene_module/scene_module_interface.hpp:263
[... 123 lines cut ...]
            z = 4.1439500000000002}, orientation = {x = 0, y = 0, 
            z = 0.68093998794346955, w = 0.73233921977424343}}, 
        longitudinal_velocity_mps = 4.16666651, lateral_velocity_mps = 0, 
        heading_rate_rps = 0, is_final = false}, 
      lane_ids = std::vector of length 1, capacity 1 = {767}}, {point = {
        pose = {position = {x = 75667.698449999996, y = 31105.475299999998, 
            z = 4.1188000000000002}, orientation = {x = 0, y = 0, 
            z = 0.6782572080599496, w = 0.734824577511342}}, 
        longitudinal_velocity_mps = 4.16666651, lateral_velocity_mps = 0, 
        heading_rate_rps = 0, is_final = false}, 
      lane_ids = std::vector of length 2, capacity 2 = {767, 764}}, {point = {
        pose = {position = {x = 75668.044550000006, y = 31109.791250000002, 
            z = 4.1712749999999996}, orientation = {x = 0, y = 0, 
            z = 0.678257208062362, w = 0.73482457750911534}}, 
--Type <RET> for more, q to quit, c to continue without paging--q
Quit
(gdb) p path->points.size()
$6 = 82
(gdb) 
```

The `stop_line_idx` is initialized here, and seems to be only suited for `path` and not `input_path`:
https://github.com/autowarefoundation/autoware.universe/blob/62476712f16cc61c0a6fc08b10f518554f1fbab0/planning/behavior_velocity_planner/src/scene_module/intersection/scene_intersection.cpp#L127-L133

A simple fix is to get rid of `input_path` and replace it by `path`, however I am not sure it is the correct solution. The code logic is beyond my understanding.

rej55, 2022-07-06:
@VRichardJP 
Thank you for your report.
I agree to replace `input_path` to `path`.

VRichardJP, 2022-07-07:
So should `input_path` be replaced everywhere by `path` or should it keep it at other places?

VRichardJP, 2022-07-12:
Yes, I tested the same modification during my test drive session and I haven't faced the issue since.
Thank you!

closed 2022-07-12 by VRichardJP