autowarefoundation/autoware_universe #1251: Timing problem in freespace_planner_node

NorahXiong, 2022-07-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In [FreespacePlannerNode::onRoute function](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/freespace_planner/src/freespace_planner/freespace_planner_node.cpp#L309), expression in null pointer check seems to be written as an assignment operation by mistake. I think the statement on the right side shows the actual intention. Otherwise the "if" statement will always get the "false" value.
`if (scenario_ = nullptr) return;`  ->  `if (scenario_ == nullptr) return;` 

However, if the statement is corrected and try to activate freespace planning in parking scenario immediately after Autoware boots, we can not get a trajectory.
[image]

### Expected behavior

Freespace planner can generate a trajectory in parking scenario as long as Autoware boots normally without any precondition.

### Actual behavior

Freespace planner can not generate a trajectory in parking scenario in the first planning task.

### Steps to reproduce

1. Correct the null pointer check statement in [FreespacePlannerNode::onRoute function](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/freespace_planner/src/freespace_planner/freespace_planner_node.cpp#L309).
2. Start planning simulator.
3. Set start pose and goal pose in a parking lot. (no trajectory will be published)
4. Move the goal pose to another place in the parking lot. (trajectories can be published)

### Versions

- OS: Ubuntu 20.04
- ROS2: Galactic
- Autoware: Autoware.universe:main

### Possible causes

In [ScenarioSelectorNode::onTimer function](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/scenario_selector/src/scenario_selector_node/scenario_selector_node.cpp#L252), scenario_selector will not publish scenario info if route is not received.
In [FreespacePlannerNode::onTimer function](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/freespace_planner/src/freespace_planner/freespace_planner_node.cpp#L409), freespace_planner will not plan unless route is received.
Mission planner publishes route only once in [goal_pose](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/mission_planner/lib/mission_planner_base.cpp#L87) or [check_point](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/mission_planner/lib/mission_planner_base.cpp#L116) callback functions.

So freespace_planner can not receive the route mission_planner published for the first time.

### Additional context

Related code is modified in PR https://github.com/autowarefoundation/autoware.universe/pull/1155.

NorahXiong, 2022-07-07:
> @NorahXiong This bug came from me. Sorry for taking your time and thank you for your investigation. Could you check this PR? #1261

I think https://github.com/autowarefoundation/autoware.universe/pull/1261 could not totally cover the problem. Actually, everything goes right with the original bug. But after correction, it will fail to generate a trajectory in parking scenario immediately after Autoware boots and this is the problem to be discussed in the issue(see #Possible causes# section). 

It is certainly reasonable of freespace_planner not receiving route before the scenario change to parking mode but the message sequence results in unexpected result. Maybe just delete this sentence or let scenario_selector send route message to freespace_planner after scenario changed to parking mode.

shmpwk, 2022-07-07:
@NorahXiong 
You're right, thank you. I've deleted nullptr check of scenario topic by https://github.com/autowarefoundation/autoware.universe/pull/1261/commits/5a4b37c88beb78ada2a3e755bb27422f7ebb5a8d.

NorahXiong, 2022-07-07:
@shmpwk By the way, [this line](https://github.com/autowarefoundation/autoware.universe/blob/7f35d5bbea9a843fcacef2de9b0d3eb39ef34948/planning/mission_planner/src/mission_planner_lanelet2/mission_planner_lanelet2.cpp#L156) blocked route visualization as the value is never set to true.

shmpwk, 2022-07-08:
@NorahXiong Exactly right, thank you! I fixed my PR.

closed 2022-07-08 by shmpwk