autowarefoundation/autoware_universe #1253: Autoware keeps following old local path after `behavior_velocity_planner` is dead

VRichardJP, 2022-07-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While I experienced the `behavior_velocity_planner` crash (described here: https://github.com/autowarefoundation/autoware.universe/issues/1247), I observed autoware did not make the vehicle stop immediately after the crash, but kept driving along the last generated local path, reached the end of the local path and kept driving straight after. Autoware only triggered an emergency stop was ~3m beyond that last point (out of path detected).

Although the out of path detection range could be reduced to have the emergency triggered earlier, it is very unsafe for autoware to keep following an old local path. For example, even though some new obstacle would be detected on the road, autoware would simply ignore them and follow the last local path it has received.

### Expected behavior

An emergency should be triggered if local paths are not generated anymore

### Actual behavior

Autoware keep driving on the old path, even after its end has been reached.

### Steps to reproduce

1. run autoware on a vehicle
2. kill `behavior_velocity_planner`

### Versions

- Ubuntu 20.04
- galactic
- main

xmfcx, 2022-07-19:
A node monitoring node should check the critical nodes and issue emergency stop if one of them crashes.

You can try to add [topic_state_monitor](https://github.com/autowarefoundation/autoware.universe/tree/main/system/topic_state_monitor) to this launch file: https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_system_launch/launch/system.launch.xml

And somehow it should issue emergency message and stop the vehicle through the [emergency_handler](https://github.com/autowarefoundation/autoware.universe/tree/main/system/emergency_handler)

mitsudome-r, 2022-08-02:
@mitsudome-r to provide additional documentation regarding emergency handler

mitsudome-r, 2022-08-30:
I've started working on this. I think I will be able to create PR this week.

mitsudome-r, 2022-09-13:
I figured out that with the default configuration, topic monitoring function is not turned on. (This probably is because it is not always safe to make a sudden stop at node failure)

If you add use_emergency handling to control module, you should be able to turn on this function and make vehicle stop when ever major planning nodes dies with unknown error.

https://github.com/autowarefoundation/autoware_launch/blob/515239b9f1d9c4161629f25c1b8e4c6ec62acfdd/autoware_launch/launch/autoware.launch.xml#L117

Modify the above launch as the following:
```
  <group if="$(var launch_control)">
    <include file="$(find-pkg-share tier4_control_launch)/launch/control.launch.py">
      <arg name="lateral_controller_mode" value="mpc_follower"/>
      <arg name="vehicle_info_param_file" value="$(find-pkg-share $(var vehicle_model)_description)/config/vehicle_info.param.yaml"/>
      <arg name="use_emergency_handling" value="true"/>
    </include>
  </group>
```

mitsudome-r, 2022-09-27:
I will close this issue once I add the above explanation in Autoware Documentation

xmfcx, 2023-01-10:
@mitsudome-r is working on https://github.com/autowarefoundation/autoware-documentation/pull/278

VRichardJP, 2023-11-04:
I have not seen the motion planner dying while autoware is driving for a while, so I am not sure if the problem still exists.

I think now there would be an emergency stop?