autowarefoundation/autoware_universe #1256: Can't launch lidar_centerpoint with error: could not load library: libcenterpoint_cuda_libraries.so.

mikechan0731, 2022-07-06:


Sharrrrk, 2022-07-18:
Can you try to rebuild the lidar_centerpoint package by removing the built targets in /build and /install, and also to make sure that the version of CUDA version in docker is same with local system.

xmfcx, 2022-07-26:
@mikechan0731 do you have any updates?

mitsudome-r, 2022-08-02:
@mikechan0731 has tried rebuilding the stack and still had the issue. However, he decided to use the default Autoware docker to avoid the issue. We will close this issue until someone else faces similar issue with his/her custom docker image.

HaoruXue, 2022-08-27:
@mitsudome-r I have the same issue with custom environment (Nvidia NGC TensorRT container with tensorrt 8.4.1-1+cuda11.6).

I fixed the issue by adding the following to `CMakeLists.txt`

```
  install(
    TARGETS
      centerpoint_cuda_lib
  )
```

You can find the commit [here](https://github.com/airacingtech/autoware.universe/commit/86a0f4fab9fd093f65f2373e8c2906af1d5ffcad).

It turns out `libcenterpoint_cuda_libraries.so` is built in the build/lidar_centerpoint folder but never installed to the install/lidar_centerpoint folder. I am curious as to why this woud work on some version of CUDA and tensorrt as it seems to be CMake issue. `tensorrt_yolo` for example as a same line that installs its built CUDA library: 

https://github.com/autowarefoundation/autoware.universe/blob/main/perception/tensorrt_yolo/CMakeLists.txt#L192

I suggest reopen this issue and see if there are other reasons causing this. Otherwise the example commit above would be the fix.

yukke42, 2022-09-21:
close this issue since this error is fixed in the PR <https://github.com/autowarefoundation/autoware.universe/pull/1916>.

closed 2022-09-21 by yukke42