autowarefoundation/autoware_universe #1311: Fail CI check build-and-test-differential (galactic, -cuda)

satoshi-ota, 2022-07-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Following CI checks fail at Build step.

- build-and-test-differential (galactic, -cuda)
- build-and-test-differential (humble, -cuda)

[image]

### Expected behavior

Fix following error.

`E: Failed to fetch https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/Packages.gz  File has unexpected size (580845 != 580040). Mirror sync in progress? [IP: 152.195.19.142 443]`

### Actual behavior

Show following error.

`E: Failed to fetch https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/Packages.gz  File has unexpected size (580845 != 580040). Mirror sync in progress? [IP: 152.195.19.142 443]`

### Steps to reproduce

Run following CI checks.

- build-and-test-differential (galactic, -cuda)
- build-and-test-differential (humble, -cuda)

### Versions

- universe: main(22/7/12)

kenji-miyake, 2022-07-12:
[Additional information]

CI link: https://github.com/autowarefoundation/autoware.universe/runs/7293172779?check_suite_focus=true

Possible causes:
```
E: Failed to fetch https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/Packages.gz  File has unexpected size (580845 != 580040). Mirror sync in progress? [IP: 152.195.19.142 443]
```

Steps to reproduce:

```console
❯ docker run --rm -it ghcr.io/autowarefoundation/autoware-universe:humble-latest-cuda
root@5b639c3d03c4:/autoware# apt update
Get:1 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease [1581 B]
Get:2 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Packages [580 kB]
Err:2 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  Packages
  File has unexpected size (580845 != 580040). Mirror sync in progress? [IP: 152.199.39.144 443]
  Hashes of expected file:
   - Filesize:580040 [weak]
   - SHA256:dc8abbaf470d3ee626f4f1f4d2871a98d8dc8f770bc592676f0d1f60637e0c2d
   - SHA1:109bfa4e5c415731aa44b1c1caae3f19754c5406 [weak]
   - MD5Sum:bf8f928cb55b1c9e4c158a69ee52b5d8 [weak]
  Release file created at: Mon, 11 Jul 2022 19:01:54 +0000
Get:3 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:4 http://packages.ros.org/ros2/ubuntu jammy InRelease [4670 B]
Get:5 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [643 kB]
Get:6 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Ign:7 https://s3.amazonaws.com/autonomoustuff-repo jammy InRelease
Get:8 https://s3.amazonaws.com/autonomoustuff-repo jammy Release [2922 B]
Get:9 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [114 kB]
Ign:10 https://s3.amazonaws.com/autonomoustuff-repo jammy Release.gpg
Get:11 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [99.8 kB]
Get:12 https://s3.amazonaws.com/autonomoustuff-repo jammy/main amd64 Packages [1363 B]
Get:13 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:14 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [226 kB]
Get:15 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:16 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:17 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
Get:18 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [4648 B]
Get:19 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [248 kB]
Get:20 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [101 kB]
Get:21 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:22 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [441 kB]
Get:23 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [292 kB]
Get:24 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [7804 B]
Get:25 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [184 kB]
Get:26 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [5798 B]
Get:27 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:28 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Fetched 24.0 MB in 4s (6112 kB/s)
Reading package lists... Done
E: Failed to fetch https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/Packages.gz  File has unexpected size (580845 != 580040). Mirror sync in progress? [IP: 152.199.39.144 443]
   Hashes of expected file:
    - Filesize:580040 [weak]
    - SHA256:dc8abbaf470d3ee626f4f1f4d2871a98d8dc8f770bc592676f0d1f60637e0c2d
    - SHA1:109bfa4e5c415731aa44b1c1caae3f19754c5406 [weak]
    - MD5Sum:bf8f928cb55b1c9e4c158a69ee52b5d8 [weak]
   Release file created at: Mon, 11 Jul 2022 19:01:54 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

kenji-miyake, 2022-07-12:
I had to remove `cuda-*` [here](https://github.com/autowarefoundation/autoware/blob/ba9d3c4e36d58cbdbd29b85d4b4f54fde001fa0c/docker/autoware-universe/Dockerfile#L41-L43).

```console
root@d9600ac0d507:/autoware# ls /etc/apt/sources.list.d/
autonomoustuff-public.list  cuda-ubuntu2004-x86_64.list  focal.list  ros2.list
```

If I remove it, `apt update` succeeds.

```console
root@d9600ac0d507:/autoware# rm /etc/apt/sources.list.d/cuda-ubuntu2004-x86_64.list
root@d9600ac0d507:/autoware# apt update
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:2 http://packages.ros.org/ros2/ubuntu jammy InRelease [4670 B]
Get:3 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Ign:4 https://s3.amazonaws.com/autonomoustuff-repo jammy InRelease
Get:5 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages [643 kB]
Get:6 https://s3.amazonaws.com/autonomoustuff-repo jammy Release [2922 B]
Ign:7 https://s3.amazonaws.com/autonomoustuff-repo jammy Release.gpg
Get:8 https://s3.amazonaws.com/autonomoustuff-repo jammy/main amd64 Packages [1363 B]
Get:9 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [101 kB]
Get:10 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [248 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [114 kB]
Get:12 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [226 kB]
Get:13 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [99.8 kB]
Get:14 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [4648 B]
Get:15 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:16 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:17 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
Get:18 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:19 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:20 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [7804 B]
Get:21 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [184 kB]
Get:22 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [441 kB]
Get:23 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [292 kB]
Get:24 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [5798 B]
Get:25 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:26 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Fetched 24.0 MB in 6s (4109 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
37 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

kenji-miyake, 2022-07-12:
I'll submit a PR to `autowarefoundation/autoware`, and recreate the Docker images.

kenji-miyake, 2022-07-12:
It wasn't resolved. I guess what we can do is only wait. :pleading_face: 
https://github.com/autowarefoundation/autoware/runs/7294080970?check_suite_focus=true#step:5:851

kenji-miyake, 2022-07-12:
Related?
- https://forums.developer.nvidia.com/t/error-during-apt-update-on-ubuntu-18-04-file-has-unexpected-size/220436
- https://forums.developer.nvidia.com/t/errors-during-apt-update-on-ubuntu2004-file-has-unexpected-size/220415
- https://forums.developer.nvidia.com/t/cuda-packages-file-for-ubuntu20-04-new-update-07-11-break-due-unexpected-file-size-580845/220419

satoshi-ota, 2022-07-12:
May I close this issue ? (Resolved ?)

satoshi-ota, 2022-07-13:
I close this issue.
@kenji-miyake Thank you for your PR :smile:

closed 2022-07-13 by satoshi-ota