autowarefoundation/autoware_universe #1325: Build process failed with "fatal error: range/v3/all.hpp: No such file or directory"

DeepakVellampalli, 2022-07-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I was following [docker installation instructions](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/#docker-with-nvidia-gpu-fails-to-start-autoware-on-arm64-devices) and encountered an error in the build process. "tier4_pcl_extensions" package failed because of missing `range/v3/all.hpp` file.

### Expected behavior

Successful build process with no failed packages. 

### Actual behavior

[image]

### Steps to reproduce

By following instructions on this [page](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/#docker-with-nvidia-gpu-fails-to-start-autoware-on-arm64-devices)

Encountered this error while trying to build the workspace inside the docker container using the following command: `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
`

### Versions

(Inside the docker)
- OS: Ubuntu 20.04
- ROS2: galactic

### Possible causes

Because of this commit: https://github.com/autowarefoundation/autoware.universe/commit/3bbbcd4a9448a8e5ef894ae8842a5e1eba10b1be where some refactoring has been done using range-v3 library.
However, that particular library doesn't seem to be found.

VRichardJP, 2022-07-12:
I encountered this too recently, I thought it was related to my custom docker build.
A temporary fix from inside the container is to run rosdep (which will find the missing library):
```
rosdep update && rosdep install -y --from-path src --ignore-src
```

angry-crab, 2022-07-13:
```
sudo apt update
sudo apt install librange-v3-dev -y
```

taikitanaka3, 2022-07-19:
@DeepakVellampalli 
Could you close this issue if you solve the problem by https://github.com/autowarefoundation/autoware.universe/issues/1325#issuecomment-1182602478.