autowarefoundation/autoware_universe #1328: planning mode doesn't change if lane change path can't be executed in behavior path planner

alanmengg, 2022-07-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In lane driving, if vehicle in lane change mode and lane change can't successfully executed such as obstacle blocks the lane change path,  it would suck with it.
[image]
[image]

### Expected behavior

if lane change can't be executed because of the obstacle, it should change to obstacle avoidance mode to plan a new path.
if planning mode change to obstacle avoidance, it can plan a new path to execute like below.
[image]

### Actual behavior

it suck with lane change path and doesn't plan new path.

### Steps to reproduce

like the picture uploaded above, put some obstacle around road,  make vehicle start and goal in different lane.

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

raminterpl, 2022-08-08:
@alanmengg : Can you please provide a map that used for testing

xmfcx, 2022-08-16:
@alanmengg could you provide map and steps to reproduce this issue?

alanmengg, 2022-08-16:
> @alanmengg could you provide the map please?
sorry, I just saw your request. here is the map 
[map.zip](https://github.com/autowarefoundation/autoware.universe/files/9349758/map.zip)

raminterpl, 2022-08-23:
Hi @alanmengg  : I confirm above issue. 
Planning mode is not changing if there is an obstacle in the path of behavior path planner.

[image]

zulfaqar-azmi-t4, 2022-08-30:
Hi @alanmengg and @raminterpl 
If I understand correctly, as lane change is approved while obstacle is blocking the lane changing path, you will not be continue. 

Since the behavior path planning is using behavior tree, one running module will be executed at a time. So if we are performing lane change, before lane change executing is complete, we will not be able to run avoidance module. Performing lane change while also avoiding obstacle is currently a limitation that is constantly under discussion internally. And modifying the behavior tree's behavior will take a bit of time for now. 

Now about the reported scenario, one thing we can do is to cancel the lane change and perform the lane changing maneuver after the path is safe. The current universe is able to do so, as shown in the below video.

https://user-images.githubusercontent.com/93502286/187413754-2ac05792-3d91-45bd-8a34-6b716bab6b13.mp4

https://user-images.githubusercontent.com/93502286/187414175-4e2db2d4-f3a6-497e-8be6-ece742ee73b7.mp4

alanmengg, 2022-09-05:
> 
thank u for your explanation.
my first thought was to change to another planning mode but it would cause other problems. but perform the lane changing maneuver in this case is  a better thing to do.

closed 2022-09-05 by alanmengg