autowarefoundation/autoware_universe #1389: planner node dies inside RCLCPP_COMPONENTS_REGISTER_NODE macro

taikitanaka3, 2022-07-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

some times planning node dies inside RCLCPP_COMPONENTS_REGISTER_NODE
this problem can be seen in 
- motion_velocity_smoother
- mpc follower
[image]

### Expected behavior

node will not die

### Actual behavior

node dies
[image]
here is back trace code
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f488e666859 in __GI_abort () at abort.c:79
#2  0x00007f488e8ef911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f488e8fb38c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f488e8fb3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f488e8fb37f in std::rethrow_exception(std::__exception_ptr::exception_ptr) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f488eba0b5d in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_t const*, void (*)()) ()
   from /opt/ros/galactic/lib/librclcpp.so
#7  0x00007f488ec234c6 in rclcpp::Service<rcl_interfaces::srv::ListParameters>::handle_request(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<void>) () from /opt/ros/galactic/lib/librclcpp.so
#8  0x00007f488eba566d in ?? () from /opt/ros/galactic/lib/librclcpp.so
#9  0x00007f488eba6014 in ?? () from /opt/ros/galactic/lib/librclcpp.so
#10 0x00007f488eba628c in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>) ()
   from /opt/ros/galactic/lib/librclcpp.so
#11 0x00007f488eba70ad in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) ()
   from /opt/ros/galactic/lib/librclcpp.so
#12 0x00007f488ebae12c in rclcpp::executors::SingleThreadedExecutor::spin() ()
   from /opt/ros/galactic/lib/librclcpp.so
#13 0x000055dc4b9044b6 in main (argc=<optimized out>, argv=<optimized out>)
--Type <RET> for more, q to quit, c to continue without paging--
    at /home/autoware/autoware.proj/build/motion_velocity_smoother/rclcpp_components/node_main_motion_velocity_smoother.cpp:56
```

registered macro code
```
// Copyright 2019 Open Source Robotics Foundation, Inc.
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <memory>
#include <string>
#include <vector>

#include "class_loader/class_loader.hpp"
#include "rclcpp/rclcpp.hpp"
#include "rclcpp_components/node_factory.hpp"
#include "rclcpp_components/node_factory_template.hpp"

#define NODE_MAIN_LOGGER_NAME "motion_velocity_smoother"

int main(int argc, char * argv[])
{
  auto args = rclcpp::init_and_remove_ros_arguments(argc, argv);
  rclcpp::Logger logger = rclcpp::get_logger(NODE_MAIN_LOGGER_NAME);
  rclcpp::executors::SingleThreadedExecutor exec;
  rclcpp::NodeOptions options;
  options.arguments(args);
  std::vector<class_loader::ClassLoader * > loaders;
  std::vector<rclcpp_components::NodeInstanceWrapper> node_wrappers;

  std::string library_name = "libmotion_velocity_smoother_node.so";
  std::string class_name = "rclcpp_components::NodeFactoryTemplate<motion_velocity_smoother::MotionVelocitySmootherNode>";

  RCLCPP_DEBUG(logger, "Load library %s", library_name.c_str());
  auto loader = new class_loader::ClassLoader(library_name);
  auto classes = loader->getAvailableClasses<rclcpp_components::NodeFactory>();
  for (auto clazz : classes) {
    std::string name = clazz.c_str();
    if (!(name.compare(class_name))) {
      RCLCPP_DEBUG(logger, "Instantiate class %s", clazz.c_str());
      auto node_factory = loader->createInstance<rclcpp_components::NodeFactory>(clazz);
      auto wrapper = node_factory->create_node_instance(options);
      auto node = wrapper.get_node_base_interface();
      node_wrappers.push_back(wrapper);
      exec.add_node(node);
    }
  }
  loaders.push_back(loader);

  exec.spin();

  for (auto wrapper : node_wrappers) {
    exec.remove_node(wrapper.get_node_base_interface());
  }
  node_wrappers.clear();

  rclcpp::shutdown();

  return 0;
}
```

### Steps to reproduce

I test with tier4 internal scenario so This description can be ambiguous for awf members
1. download scenario [here](https://evaluation.tier4.jp/evaluation/reports/b5703a98-695f-518a-accd-2a43eda48030/tests/ad2e8836-c93d-534c-a736-d44b10b58918/8e26cf30-261f-5df4-87b7-5fb910c875d9/94d581cc-46b8-591e-a99e-0774fe17ab9a?project_id=prd_jt) 
2. test lane change scenario 
3. see if motion velocity smoother dies.
```
[openscenario_interpreter_node-3] [ERROR] [motion_velocity_smoother-46]: process has died [pid 574, exit code -6, cmd '/home/autoware/autoware.proj/install/motion_velocity_smoother/lib/motion_velocity_smoother/motion_velocity_smoother --ros-args -r __node:=motion_velocity_smoother -r __ns:=/planning/scenario_planning --params-file /tmp/launch_params_q0vycl5e --params-file /home/autoware/autoware.proj/install/planning_launch/share/planning_launch/config/scenario_planning/common/common.param.yaml --params-file /home/autoware/autoware.proj/install/planning_launch/share/planning_launch/config/scenario_planning/common/motion_velocity_smoother/motion_velocity_smoother.param.yaml --params-file /home/autoware/autoware.proj/install/planning_launch/share/planning_launch/config/scenario_planning/common/motion_velocity_smoother/JerkFiltered.param.yaml --params-file /tmp/launch_params_tiz0lsed --params-file /tmp/launch_params_q5f5vl_g -r ~/input/trajectory:=/planning/scenario_planning/scenario_selector/trajectory -r ~/output/trajectory:=/planning/scenario_planning/trajectory -r ~/input/trajectory:=/planning/scenario_planning/scenario_selector/trajectory -r ~/output/trajectory:=/planning/scenario_planning/scenario_selector/trajectory -r ~/input/external_velocity_limit_mps:=/planning/scenario_planning/max_velocity -r ~/output/current_velocity_limit_mps:=/planning/scenario_planning/current_max_velocity'].
```

### Versions

awf-latest

### Possible causes

this sometimes happen while running CI

mitsudome-r, 2022-09-13:
@taikitanaka3 Is anyone from TIER IV working on this? Since the scenario file to reproduce the error is only accessible by TIER IV, we cannot assign anyone from AWF.

taikitanaka3, 2022-09-13:
@mitsudome-r 
computing team is currently investigating related to this problem.
But since it's rare case to die node during run time, it's hard without scenario simulator.

mitsudome-r, 2022-09-27:
I have removed high-priority tag from this issue since we cannot assign anyone unless there is a way to reproduce with publicly available environment. We can bring it to attention once it is reproducable.

xmfcx, 2023-03-07:
@taikitanaka3 is it possible to reproduce this on the local machine too?

taikitanaka3, 2023-03-08:
@xmfcx 
sorry this might be CI unique issue so it's a bit difficult to reproduce in local machine.
And I close this issue because this package is old enough, if someone can reproduce similar issue I will reopen this.

closed 2023-03-08 by taikitanaka3