autowarefoundation/autoware_universe #1418: `obstacle_stop_planner` makes the vehicle stop for a far away obstacle

VRichardJP, 2022-07-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

During a self driving test, I observed our vehicle would stop for a brief moment in the middle of an intersection, always at the same spot. From Rviz view, autoware seem to stop the vehicle when an "obstacle on the path" wall appears. What is interesting is that the wall appears for a "far away" obstacle (relatively to the path).  I don't have any screenshot of the incident, but the situation looks like so:

[image]

The vehicle comes from the bottom and goes up. After around 200 meters, the path comes back to the intersection and crosses from right to left. The default planning range is 300 meters, so autoware is already planning what to do when the vehicle comes back to the intersection from the right. As expected, the `obstacle_stop_planner` scene module creates a stop point for when the vehicle comes back from the right. However, this stop point seems to make the vehicle stop when it goes straight from the bottom.

The wall clearly says "obstacle on the path", which is only created by `obstacle_stop_planner`:
https://github.com/autowarefoundation/autoware.universe/blob/16228926f9baa33df9ee5398474d6577528c4757/planning/obstacle_stop_planner/src/debug_marker.cpp#L180

The problem disappeared after I reduced autoware's planning range from 300 to 100 meters (the right->left only comes after ~200m)

### Expected behavior

The vehicle should not be stopped for a stop point very far in the future.

### Actual behavior

The vehicle stops.

### Steps to reproduce

1. Use a lanelet map where the vehicle goes through the same intersection twice (like a 8-shaped loop)
2. Have an obstacle blocking the path right after the second crossing (like on the picture)

### Versions

- Ubuntu 20.04
- ROS2 galactic
- main

### Possible causes

I guess that somewhere autoware decide to stop the vehicle because it is "close" to the stop point, whereas the stop point is actually very far if we consider the distance relatively to the path.

xmfcx, 2022-07-26:
@VRichardJP can you share a vector map for us to reproduce this on?

VRichardJP, 2022-07-27:
Here is a simplified version of the lanelet map: [lanelet2_map.zip](https://github.com/autowarefoundation/autoware.universe/files/9194131/lanelet2_map.zip)

I have increased [`forward_path_length`](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/config/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/behavior_path_planner.param.yaml) to 400.0 to make it easier to visualize.

And the setup looks basically like so:
[image]

mitsudome-r, 2022-08-02:
@mitsudome-r to test this with planning_simulator this week.

mitsudome-r, 2022-10-04:
@VRichardJP I've tested with the simulator and took the video. Could you confirm if this is different from what you see? It seems the vehicle is not stopping for the pedestrian that is far ahead in the path and seems to be doing a natural maneuver.
https://www.youtube.com/watch?v=-EVmKwGdsog

VRichardJP, 2022-10-05:
@mitsudome-r Thank you for the effort. I understand the issue is maybe not reproducible with the simulator. 
I don't know the next time I will have access to the site. When I do, I will try to reproduce the issue and record some rosbag.

closed 2022-10-05 by VRichardJP