autowarefoundation/autoware_universe #1420: Speed profile in incorrect in slopes

VRichardJP, 2022-07-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

On our site, there is a small slope going up and down. For both cases, the lanelet speed limit is set to 10km/h.
When the vehicle climbs the slope, the speed seems to be stuck at around 7km/h. When the vehicle goes down the slope, the speed goes to 14km/h. There is no such problem on flat road.

### Expected behavior

Whether it goes up or down, the vehicle should be able to reach and keep the 10 km/h.

### Actual behavior

The vehicle can't reach the max speed while going up, and drives faster than max speed when going down

### Steps to reproduce

Drive on slope

### Versions

- Ubuntu 20.04
- galactic
- main

### Possible causes

accel/brake map values have been calibrated on flat road. Does autoware pushes the accel/brake pedals the same in slope and flat road?

maxime-clem, 2022-07-26:
There is an implementation of slope compensation in the current longitudinal controller:
- https://github.com/autowarefoundation/autoware.universe/blob/main/control/trajectory_follower_nodes/design/longitudinal_controller-design.md#slope-compensation
- https://github.com/autowarefoundation/autoware.universe/blob/main/control/trajectory_follower/design/trajectory_follower-pid-design.md#slope-compensation

It is deactivated by default (https://github.com/autowarefoundation/autoware.universe/blob/main/control/trajectory_follower_nodes/param/longitudinal_controller_defaults.param.yaml#L8) and I am not sure if it has ever been tested on a real vehicle.

VRichardJP, 2022-07-26:
Thank you! I will try that parameter.

I remember I did not have the issue with architecture proposal, but i see the same parameter used to be set to true: https://github.com/tier4/autoware_launch/blob/2d6815ea222deb4bfebc7c4d54693eee35c8add6/control_launch/config/velocity_controller/velocity_controller_param.yaml#L42

TakaHoribe, 2022-07-27:
@VRichardJP @maxime-clem The slope compensation was developed in the very old version (Architecture Proposal) and fully tested in our (TIER IV) vehicles.
Remark: you need to check the source of the slope value carefully. If you use the `pitch` angle for the slope computation, make sure the pitch is not osscilated. Also note that the pitch angle does not perfectly reflect the slope angle, it is road slope + vehicle mechanical pitch osscilation + localization noise. If you use the `trajectory_z_pos` for the slope computation, make sure the `z` value on your map is well defined.

xmfcx, 2022-08-09:
@VRichardJP Is this issue resolved? Are there any updates on this issue?

VRichardJP, 2022-08-10:
@xmfcx I have not had the chance to test the slope compensation on site yet. At least the cause of my issue has been identified. Maybe we can close this issue, and I'll open a new one if I there is a problem with `enable_slope_compensation` option (when I get the chance to test it.)

xmfcx, 2022-08-16:
@VRichardJP Ok I will close this issue then.

closed 2022-08-16 by xmfcx