autowarefoundation/autoware_universe #1426: Error accumulation in EKF localizer

meliketanrikulu, 2022-07-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

We tested localization in vehicle and we observe ndt aligned point cloud and concatenated point cloud. After moving for a while, we saw that the ndt align point cloud is not broken, but the concatanated point cloud is not match with the map.We think that ekf is not working properly because the concatenated point cloud is localized relative to ekf.

### Expected behavior

We expect the error not to rise in our localization output with Ekf localizer.

### Actual behavior

[Here](https://www.youtube.com/watch?v=5SPJWTWwieQ) is our video link. where the light pink point cloud represents the NDT align point cloud and the colored point cloud is the concataneted point cloud

### Steps to reproduce

[Test Bags](https://drive.google.com/drive/folders/1gqQYP_vg03bd5irsSyq4-Jp9ingyxzMd?usp=sharing)
You can reproduce ndt align point cloud when NDT works.
[Maps](https://drive.google.com/file/d/1CLJ9acywX_HgM0OAnhoSJs9V8gR82srD/view)
We are using default parameters and our yaw bias estimation is false. 

### Possible causes

We think that ekf is not working properly because the concatenated point cloud is localized relative to ekf.

YamatoAndo, 2022-07-26:
@meliketanrikulu 
Probably because the yaw angle of the TF(base_link<->lidar sensor) is misaligned.

You can see the yaw angle error with the following command.
`ros2 topic echo /localization/pose_twist_fusion_filter/ekf_localizer/estimated_yaw_bias`

I think this problem can be solved by adjusting TF using the yaw angle error.

meliketanrikulu, 2022-07-27:
> Probably because the yaw angle of the TF(base_link<->lidar sensor) is misaligned.

Thanks for your reply. I don't think it's a bug with the base_link-lidar calibration issue. Because initially, when the vehicle is stationary, concatenated point cloud and ndt align cloud are matching. After a while, the difference begins to open. While moving, the difference increasing and the difference does not close when the vehicle stops again.

When I looked at the messages from this topic (-->  /localization/pose_twist_fusion_filter/ekf_localizer/estimated_yaw_bias), I saw that the values were low at first and then increased. I will share a video about it with you.

YamatoAndo, 2022-07-27:
The ekf_localizer estimates the bias of the yaw angle, and updates the bias while driving.
Therefore, it should behave exactly as you said.

meliketanrikulu, 2022-08-09:
> The ekf_localizer estimates the bias of the yaw angle, and updates the bias while driving. Therefore, it should behave exactly as you said.

Hi @YamatoAndo . You are right this problem related to base_link - lidar calibration. Thanks for your help.

closed 2022-08-09 by meliketanrikulu